David-OConnor/stm32-hal #1: desired features

xoviat, 2021-03-08:
This is a todo list of what features I'd like to see in this to consider using it; not that my opinion carries much weight, but it's a good place to start.

- [ ] QEI encoder read/reset/enable interrupt
- [ ] DMA Serial Read/Write with nb::WouldBlock API; interrupt and one-at-a-time API is simplest to implement but not useful
- [ ] DMA I2C Read/Write with nb::WouldBlock
- [ ] DMA SPI Read/Write with nb::WouldBlock
- [ ] DMA ADC Read/Write with nb::WouldBlock
- [ ] bxcan Implementaiton
- [ ] ExtiPin
- [ ] DAC
- [ ] USB

There are nice to have, but not critical:

- [ ] RTC
- [ ] Crc32
- [ ] DMA2D

IMO, DMA implementations should be the default and should be implemented first. They are the most difficult to implement, but also the most useful.

David-OConnor, 2021-03-08:
DAC and RTC are done. (For f3 and L4)

I plan to start with these, since they should be quick to knock out, and I can test them on current projects:

- GPIO, including EXTI. I plan to take a significant break from the existing approaches, by moving away from type state programming. It complicates the multi-family aspect of this, but the main reason is it makes passing things across function boundaries, or declaring them statically untenable. Thoughts? The downside is it won't catch misconfigured pins at compile time.

Essentially, I think expecting the user to setup the pins correctly (eg cross-check the user manual, and/or copy+paste examples from the examples folder) is worth avoiding this:

```rust
    Sensor<
        I2c<
            I2C1,
            (
                PB6<Alternate<AF4, Output<OpenDrain>>>,
                PB7<Alternate<AF4, Output<OpenDrain>>>,
            ),
        >,
        ic::Mlx90614,
    >
```

- SPI (Probably based on H7xx, but open to suggestions)
- low power (Based on L4/F3 forks I've been using)

Could you help with the DMA? I haven't used it before, so don't have a grasp on where to start.

Would also like to start implementing and eventually hardware testing the current modules with H7 and L5.

xoviat, 2021-03-08:
> GPIO, including EXTI. I plan to take a significant break from the existing approaches, by moving away from type state programming. It makes passing things across function boundaries, or declaring them statically tough. Thoughts? The downside is it won't catch misconfigured or wrong pins at compile time.

I can take a look at what your proposal is, but I am personally in favor of type-state programming.

David-OConnor, 2021-03-08:
It's a big departure, and may not be worth it, but I'll still probably go with it at first. Here are some thoughts, as they apply to GPIO:

Big:
- With TST, using the wrong pin, or setting it up wrong will give you a compile error. Without you'll get unexpected runtime behavior (like the peripheral not working)
- TST leads to unwieldy syntax when types are explicitly declared (see example above). This includes global constants, functions, and struct fields. Applies upstream, ie to devices that use a bus. Can ease the pain through documenting every peripheral (eg I2C, SPI) with example function signatures. Rust docs doesn't do a great job auto-generating these, so should document there as well. Still would be verbose.

Small:
- It feels strange to pass arguments (ie pin structs) that aren't used directly.
- TST will complicate the multi-family aspect of this crate, requiring more feature gates on the bus and GPIO modules. That said, that comes with the territory, and wouldn't be too bad, since these are similar across families. Ie we can expect PB6 and PB7 to be I2C pins on most or all devices.

xoviat, 2021-03-08:
So from the chat, the best approach might be to:

- Use `new<T>() -> Self`
- Add `new_unchecked() -> Self` to all classes with `new()` that require pins
- Remove `free()` from all classes

David-OConnor, 2021-03-08:
Agree. Going to start with `new_unchecked()`, since we can do this before making the GPIO module, and it's easier to implement cross-family.

David-OConnor, 2021-03-09:
`new_unchecked` add for dac and i2c.

David-OConnor, 2021-03-09:
Would you be willing to PR the DMA stuff once I have SPI, I2C, SERIAL, and ADC working on a few families?

xoviat, 2021-03-09:
Yes.

Dirbaio, 2021-03-10:
> DMA xxx with nb::WouldBlock

It is *completely impossible* to implement `nb` APIs with DMA.

Consider an API like this:

```rust
fn read(&mut self, bytes: &mut [u8]) -> nb::Result<(), Self::Error> { ..  }
```

On first call, this would start the DMA read into `bytes`, and then return `WouldBlock`. The problem is after return, the `bytes` buffer is *no longer borrowed*. The user could read from it which would race with DMA writes. Or *even deallocate it*, which would cause DMA to overwrite arbitrary memory.

xoviat, 2021-03-10:
Not true. See how it's done in the f7 HAL.

Dirbaio, 2021-03-10:
The f7 hal doesn't implement the embedded-hal nonblocking traits with DMA, does it?

David-OConnor, 2021-04-18:
Hey - DMA is over my head for now, so I don't imagine solving this issue in the near future. If someone knowledgeable with DMA want to give it a go, please do! (I'll get to QEI and CAN, also mentioned in this issue, sooner)

David-OConnor, 2021-05-11:
Going to tackle DMA, but don't plan to base it off of `nb`.

David-OConnor, 2021-05-19:
Could you provide more info on QEI? Can't find it in the RMs.

David-OConnor, 2021-05-24:
Re-open if you have info on QEI, or would like to tackle DMA2D.

closed 2021-05-24 by David-OConnor

kellda, 2022-07-05:
Re QEI: It seems that on STM32s, QEIs are just a special mode for timers. See e.g. “29.4.18 Encoder interface mode” in https://www.st.com/resource/en/reference_manual/rm0440-stm32g4-series-advanced-armbased-32bit-mcus-stmicroelectronics.pdf.