David-OConnor/stm32-hal #6: Feature gate more of the peripherals

theunkn0wn1, 2021-03-16:
I decided to look into introducing F4 support.

the F4 does not have `TIM15`,`TIM16`. FEature gating will be necessary for `src.timer.rs`'s include on line 9.

The DAC has a different name, but feature gating exists here so that can be addressed directly.

David-OConnor, 2021-03-16:
Re timers: Across families, the functionality (including what a timer of a given name can do) is similar or identical. But, each variant has its own combination of timers it uses, so that module is loaded with feature-gating on the imports and macro-calls.

theunkn0wn1, 2021-03-16:
Yep, the issue is `TIM15` and `TIM16` specifically are not feature gated, and don't exist on the F446 (and likely others in the F4 family; would need to acquire additional reference manuals).

David-OConnor, 2021-03-16:
Oh weird! For a while I thought TIM1 was universal too.

PR #8 (2021-03-16): Feature gate TIM15 and TIM16
resolves #6
commit 7b5526f6 (2021-03-16): [#6] feature gate TIM15 and TIM16
resolves #6
closed 2021-03-16 by David-OConnor