David-OConnor/stm32-hal #9: Error: RTT not supported on device without HW breakpoints

jonahbron, 2021-04-20:
My project is just started.  I followed the instructions for stm32-hal2 and the Knurling template.  The project builds and flashes to the device successfully as far as I can tell.  However, I get the titular error about RTT support.  Here's the full output:

```bash
> cargo rb hello                                                                                                                                                                          
    Finished dev [optimized + debuginfo] target(s) in 0.01s
     Running `probe-run --chip STM32L562CETxP target/thumbv7m-none-eabi/debug/hello`
  (HOST) INFO  flashing program (6.98 KiB)
  (HOST) INFO  success!
Error: RTT not supported on device without HW breakpoints
```

`hello` is provided by the app template.  It's not clear whether this error is coming from the HAL or probe-run or probe-rs, so I'm starting here.  Is it possible to simply turn off RTT?

I've uploaded my project here:
https://github.com/jonahbron/rollova/

David-OConnor, 2021-04-20:
I'll look into this and get back to you. Are you running this with the thumbv8 target?

David-OConnor, 2021-04-20:
This target: `thumbv8m.main-none-eabihf`

David-OConnor, 2021-04-20:
Updated the example `config.toml` with which target to use by MCU, to make this more explicit. (I don't think this is the problem here of the RTT error)

David-OConnor, 2021-04-20:
Note: To narrow it down to see if this crate is the cause, try commenting out the `stm32_hal2` import, and all uses of it in your code, and see if you get the same error

jonahbron, 2021-04-21:
Thanks for the quick response.  I mis-understood the target options.  Unfortunately, changing the target had no effect on the error.

Removing the dependency and references in the code also had no effect.  Do you think it's a probe-rs issue then?

David-OConnor, 2021-04-21:
I think it's likely with `probe-rs` or `probe-run`, with a small chance it's `defmt`. I'm curious what's going on, and would like to put a fix for it in the readme here etc once we figure it out.

jonahbron, 2021-04-21:
Yes, ST-LINK (v2).  Only connecting 4 wires (Vdd, Vcc, SWDIO, SWCLK).  I've used OpenOCD to read from the registers, but that's it.  I'm not certain what exactly I would replace with OpenOCD, but I can look into it if you think that might help.

David-OConnor, 2021-04-21:
Are you able to connect the reset line as well? I'm not sure if that's related, as when I disconnect it, probe-rs can't recognize the device at all.

jonahbron, 2021-04-21:
Just tried with the RST line connected, but the result was the same.

Okay, let me do some research and figure out how to do that.

jonahbron, 2021-04-21:
All right, I made a lot of progress, but didn't quite get there.  After going through everything in the [installation section](https://docs.rust-embedded.org/book/intro/install.html) of the Embedded Rust Book, I got most of the way through the GDB [debugging section](https://docs.rust-embedded.org/book/start/hardware.html#debugging).  I could run `load` over gdb, but if I `continue` to run the program, it spits this out:

```
(gdb) continue
Continuing.

Program received signal SIGTRAP, Trace/breakpoint trap.
lib::__bkpt () at asm/lib.rs:49
49	asm/lib.rs: No such file or directory.
```

This seems kind of unrelated to the previous error about RTT though.  I can set breakpoints as the book suggests, and they seem to work fine.

David-OConnor, 2021-04-21:
If you haven't already, I recommend posting issues in the probe-rs and/or probe-run repos. I'm curious what the cause is. Wish I had an L5 to try to reproduce with.

David-OConnor, 2021-04-22:
Another thought: do you have the `l5 rt` feature enabled in `Cargo.toml` for this HAL?
ie:
`stm32-hal2 = { path = "../../stm32-hal", features = ["l562", "l5rt"]}`

jonahbron, 2021-04-22:
I will file a bug over there next, but I think I've gotten closer to finding the source of the openocd issue.

```
>>> arm-none-eabi-gdb -q target/thumbv8m.main-none-eabihf/debug/hello
Reading symbols from target/thumbv8m.main-none-eabihf/debug/hello...
(gdb) target remote :3333
Remote debugging using :3333
cortex_m_rt::Reset ()
    at /home/jonah/.cargo/registry/src/github.com-1ecc6299db9ec823/cortex-m-rt-0.6.13/src/lib.rs:497
497	pub unsafe extern "C" fn Reset() -> ! {
(gdb) load
Loading section .vector_table, size 0x400 lma 0x8000000
Loading section .text, size 0x1410 lma 0x8000400
Loading section .rodata, size 0x448 lma 0x8001810
Loading section .data, size 0x30 lma 0x8001c58
Start address 0x08000400, load size 7304
Transfer rate: 17 KB/sec, 1826 bytes/write.
(gdb) monitor arm semihosting enable
semihosting is enabled

(gdb) break main
Breakpoint 1 at 0x80004ae: file /home/jonah/Projects/rollova/src/bin/hello.rs, line 6.
Note: automatically using hardware breakpoints for read-only addresses.
(gdb) continue
Continuing.

Breakpoint 1, hello::__cortex_m_rt_main_trampoline ()
    at /home/jonah/Projects/rollova/src/bin/hello.rs:6
6	#[cortex_m_rt::entry]
(gdb) step
halted: PC: 0x080004b4
hello::__cortex_m_rt_main () at /home/jonah/Projects/rollova/src/bin/hello.rs:8
8	    let x = 1 + 3;
(gdb) step
halted: PC: 0x080004be
halted: PC: 0x080004c0
9	    defmt::info!("Hello, world!, {}", x);
(gdb) step
halted: PC: 0x080008e4
defmt::export::acquire ()
    at /home/jonah/.cargo/registry/src/github.com-1ecc6299db9ec823/defmt-0.2.1/src/export.rs:36
36	    unsafe { _defmt_acquire() }
(gdb) step
halted: PC: 0x08000778
defmt_rtt::_defmt_acquire ()
    at /home/jonah/.cargo/registry/src/github.com-1ecc6299db9ec823/defmt-rtt-0.2.0/src/lib.rs:25
25	#[defmt::global_logger]
(gdb) step
defmt_rtt::{{impl}}::acquire ()
    at /home/jonah/.cargo/registry/src/github.com-1ecc6299db9ec823/defmt-rtt-0.2.0/src/lib.rs:39
39	        let primask = register::primask::read();
(gdb) step
cortex_m::register::primask::read ()
    at /home/jonah/.cargo/registry/src/github.com-1ecc6299db9ec823/cortex-m-0.7.2/src/register/primask.rs:29
29	    let r: u32 = call_asm!(__primask_r() -> u32);
(gdb) step
halted: PC: 0x08000b80
lib::__primask_r () at asm/lib.rs:50
50	asm/lib.rs: No such file or directory.
(gdb) 
```

As you can see, stepping through the program makes it all the way to a crate called cortex-m, which is a dependency of stm32-hal2.  Some missing assembly it seems.  Does that spark any ideas?

David-OConnor, 2021-04-22:
Have you confirmed you're using the `l5rt` feature when loading the HAL? This triggers adding the runtime feature in that cortex-m crate.

jonahbron, 2021-04-23:
I misunderstood the example and didn't realize both were needed.  I've updated it, but the behavior is not different (even after a `cargo clean` and re-building).  Latest is pushed up to the repo.

https://github.com/jonahbron/rollova/

David-OConnor, 2021-05-12:
Any progress?

David-OConnor, 2021-05-18:
Going to close this, since it's likely not something we can fix here. Please re-open, or post another issue if you find out more.

closed 2021-05-18 by David-OConnor