David-OConnor/stm32-hal #12: cargo build error

ljalyn0820, 2021-06-10:
..\stm32-hal2-0.2.8\src\timer.rs:319:13
319 |             pub fn countdown(&self) -> u32 {
    |             |
    |             duplicate definitions for `countdown`
    |             other definition for `countdown`
All pub fn void reported errors.

David-OConnor, 2021-06-10:
Something feels off about this. GH bot?

ljalyn0820, 2021-06-11:
hi, 
 In the  Cargo.toml, stm32-hal2 = { version = "^0.2.8", features = ["f4", "stm32f4"]},when i cargo build,
error：
..\stm32-hal2-0.2.8\src\timer.rs:319:13
319 | pub fn countdown(&self) -> u32 {
| |
| duplicate definitions for countdown
| other definition for countdown
All pub fn void reported errors.
error[E0592]:duplicate definitions with name `set_prescaler`

David-OConnor, 2021-06-11:
Hi! This is a documentation issue. The available features for f4 are shown in Cargo.toml here: https://github.com/David-OConnor/stm32-hal/blob/main/Cargo.toml#L83

So, the line in Cargo.toml should be something like this:
`stm32-hal2 = { version = "^0.2.8", features = ["f405", "f4rt"]}`

There's a line that's supposed to throw a compile error in `lib.rs`, but it's not working properly. I'll update the Readme to provide guidance on this. Thanks for bringing it up - let me know if you still have issues after making this change.

ljalyn0820, 2021-06-11:
Hi,I see. Thank you for your help！

David-OConnor, 2021-06-11:
Readme updated to be more clear

ljalyn0820, 2021-06-11:
OK,Thanks!

closed 2021-06-11 by ljalyn0820