David-OConnor/stm32-hal #14: STM32F334 timer issues

alianse777, 2021-07-10:
I tried uploading examples/timer.rs and it seems that duty cycle is always full, not matter if it is set to 1.0 or 0.0 or with set_duty (tested with led), but I can toggle it with normal gpio output.

David-OConnor, 2021-07-10:
Thanks for the report. I'll fix the example error, and figure out why the duty isn't setting.

Are you able to set the duty directly using the PAC, on a timer set up by the HAL?

David-OConnor, 2021-07-10:
Using just this code (modified for the timer and output pin you're using), what kind of output do you get? What happens if you vary the final argument?

```rust
    let _pwm_pin = gpioa.new_pin(0, PinMode::Alt(AltFn::Af1));

    let mut pwm_timer = Timer::new_tim2(dp.TIM2, 2_400., &clock_cfg, &mut dp.RCC);
    pwm_timer.enable_pwm_output(TimChannel::C1, OutputCompare::Pwm1, CountDir::Up, 0.5);
    pwm_timer.enable();
```

alianse777, 2021-07-11:
I tried setting it between 0.0 or 1.0, still full duty. (I use timer 1)
Here is my code:
let mut dp = pac::Peripherals::take().unwrap();
let clock_cfg = Clocks::default();
clock_cfg.setup(&mut dp.RCC, &mut dp.FLASH).unwrap();
let _a_pwm = gpioa.new_pin(8, PinMode::Alt(AltFn::Af6));
let mut pwm_timer = Timer::new_tim1(dp.TIM1, 20000., &clock_cfg, &mut dp.RCC);
pwm_timer.enable_pwm_output(TimChannel::C1, OutputCompare::Pwm1, CountDir::Up, 1.0); // PA8
pwm_timer.enable()

stm32-hal added as git dependency with latest commit.

David-OConnor, 2021-07-11:
What do you mean specifically by "full duty"? Does choosing different frequencies work?

alianse777, 2021-07-11:
I mean the led is fully bright no matter if dc is set to 0.0 or 1.0 or 0.5.  I cannot measure waveform. And no, choosing different frequencies gives the same result.

alianse777, 2021-07-11:
Anyway, tim1 works in stm32f3xx-hal so worth checking out for implementation details.

David-OConnor, 2021-07-11:
Interesting. This could be related to Tim1 being an Advanced Control timer vice a General purpose timer. I'll let you know once I sort it out.

toudi, 2021-08-16:
not sure if my comment would be of any help here but I also started doing some timers (in order to learn myself the interrupts) and I've also started on timer.rs from the examples.

anyhow, because the example used TIM3, so did I:

```rust
let mut timer = Timer::new_tim3(dp.TIM3, 0.5, &clock_cfg);
timer.enable_interrupt(TimerInterrupt::Update); // Enable update event interrupts.
timer.enable();
```

then in the interrupt:

```rust
free(|_cs| {
    // Clear the interrupt flag. If you ommit this, it will fire repeatedly.
    unsafe { (*pac::TIM3::ptr()).sr.modify(|_, w| w.uif().set_bit()) }
    // the above line seems to have no effect - the interrupt *does* fire repeatedly
    defmt::info!("2 seconds have passed");
});
```

then I tried to follow the comment about mutex and ended up with the following code:

```rust
use stm32_hal2::pac::TIM3;

static TIM3M: Mutex<RefCell<Option<Timer<TIM3>>>> = Mutex::new(RefCell::new(None));
// wow, that looks scary

fn main() -> ! {
    // ...
    free(|cs| {
        TIM3M.borrow(cs).replace(Some(timer));
    });
    // ...
}

#[interrupt]
/// Timer interrupt handler; runs when the countdown period expires.
fn TIM3() {
    free(|_cs| {
        let mut timer_mutex = TIM3M.borrow(cs).borrow_mut();
        let timer = timer_mutex.as_mut().unwrap();
        timer.clear_interrupt(TimerInterrupt::Update);

        defmt::info!("2 seconds have passed");
    });
}
```

et voila! now it works :)

David-OConnor, 2021-08-16:
Good catch - those 2 bits of code were intended to be equivalent. (use the PAC directly with a raw pointer to clear the flag, and storing the timer struct globally, and using its `clear_interrupt` method. It turns out there's a typo in the example: `uif().clear_bit()` is required to clear the flag.

Note that this is a trap: There's similar functionality for several other peripherals, and a lot of them use `set_bit()` and an interrupt-clear register, or will do one or the other depending on stm32 variant. (`clear_interrupt` methods should pick the right one based on MCU)

Examples will be fixed next commit.

toudi, 2021-08-16:
@David-OConnor you're absolutely correct! this indeed makes the code work again. Not sure if this helps OP's question

toudi, 2021-08-16:
sorry, one more thing. I was also thinking that since the dev board uses a single core, there's little sense in introducing a mutex as it would only slow things down, right? so would it be OK to implement a "fake" mutex that doesn't actually lock anything but only makes the code compliant with rust requirement? Or is it just the wrong way of doing things and one should simply use the `clear_bit` thing?

David-OConnor, 2021-08-16:
You can still have race conditions with a single core, generally caused by interrupts. Eg you're performing a task, and a higher-priority interrupt triggers something else that uses the same resource.

Generally, we prevent race conditions by using Mutexes, critical sections, and atomics. (Eg some registers support atomic writes, and there are `core::sync atomic` variables available.

David-OConnor, 2021-11-04:
@alianse777 : any luck?

David-OConnor, 2021-11-23:
Closing for now due to inactivity. Feel free to re-open A/R.

closed 2021-11-23 by David-OConnor