David-OConnor/stm32-hal #16: problem with writing clock information to the RCC [f401 and f411]

toudi, 2021-08-05:
Hello :) I've decided to create an issue and stop spamming you with email directly as I've made some progress. First of all, I would like to thank you for the wonderful work you've done with this project.

Just for reference, I'm testing this code with the following black pill boards:

- STM32F401CEUx
- STM32F411CEUx

Also again, for reference, this is my second hello-world with rust (I just started to learn the language) so if any of my questions feel dumb please forgive me.

here's my minimal example:

```
fn main() -> ! {
    let cp = cortex_m::Peripherals::take().unwrap();
    let mut dp = pac::Peripherals::take().unwrap();

    stm32_hal2::debug_workaround();

    let clock_cfg = Clocks::default();

    // as soon as I uncomment this line, things go south
    // clock_cfg.setup(&mut dp.RCC, &mut dp.FLASH).unwrap();

    // Setup a delay, based on the Cortex-m systick.
    let mut delay = Delay::new(cp.SYST, clock_cfg.systick());
    let mut led = Pin::new(Port::C, 13, PinMode::Output);

    loop {
        defmt::info!("Program is alive");
        led.set_low(); // I think the LED is activated on 0
        delay.delay_ms(1_000_u32);
        led.set_high();
    }
}

```

so there are two things happening. First of all, I only see the message once. This part I kind of understand, as without the clocks being set, the board doesn't actually know how much is 1000 ms. 

Obviously the most interesting part is about the clock_cfg.setup(). As soon as I leave the line as it is, the flasher simply fails to start the program (it flashes OK) which is indicated by the following output:

```
(HOST) INFO  flashing program (7.95 KiB)
(HOST) INFO  success!
(HOST) ERROR Max number of RTT attach retries exceeded.
Error: RTT control block not found in target memory.
```

so essentially what I think is happening is that something inside this function (clock_cfg.setup()) throws an exception and rust goes to panic. As indicated in the docstring here:

```
// same panicking *behavior* as `panic-probe` but doesn't print a panic message
// this prevents the panic message being printed *twice* when `defmt::panic` is invoked
#[defmt::panic_handler]
fn panic() -> ! {
    cortex_m::asm::udf()
}
```

the panic message is skipped so that's why I think I can't see anything. I have then tried the following trick:

```
let setup = clock_cfg.setup(&mut dp.RCC, &mut dp.FLASH);
match setup {
    Ok(_) => defmt::info!("setup result succesful"),
    Err(e) => defmt::info!("Error in setup:"),
}
```

but this, again, fails (produces the same message about RTT).

I've also tried to setup panic-probe with different features set without any progress.

what can I do to investigate deeper?

again, thank you for the wonderful work. I absolutely love the syntax!

David-OConnor, 2021-08-06:
Hey - thanks for the report. Before it was tough to diagnose, but at this point, it's almost certainly a bug with this library's F4 clock config. I don't need any more info at this time, and will keep you posted.

David-OConnor, 2021-08-06:
The default config is setting a speed too high for the PLL for F401 and F411. What happens if you try this clock code?

```rust

let clock_cfg = Clocks {
    plln: 84,
    ..Default::default();
};

clock_cfg.setup(&mut dp.RCC, &mut dp.FLASH).unwrap();
```

toudi, 2021-08-06:
ok, there's a huge progress now :D when I did the first solution (i..e `plln: 84`), it panicked but this time I could actually see some output:

```
ERROR panicked at 'called `Result::unwrap()` on an `Err` value: SpeedError', src/main.rs:41:49
────────────────────────────────────────────────────────────────────────────────
stack backtrace:
   0: HardFaultTrampoline
      <exception entry>
   1: lib::inline::__udf
        at ./asm/inline.rs:172:5
   2: __udf
        at ./asm/lib.rs:49:17
   3: _ZN8cortex_m3asm3udf17he373a17b1a318e8eE
   4: rust_begin_unwind
   5: core::panicking::panic_fmt
        at /rustc/a178d0322ce20e33eac124758e837cbd80a6f633/library/core/src/panicking.rs:92:14
   6: core::result::unwrap_failed
        at /rustc/a178d0322ce20e33eac124758e837cbd80a6f633/library/core/src/result.rs:1355:5
   7: _ZN11rusty_blink18__cortex_m_rt_main17h90e745ed8fa4394aE
   8: main
   9: ResetTrampoline
  10: Reset
(HOST) ERROR the program panicked
```

now with the second solution everything seems to be working (at least the code is waiting each second and I can see the output), however:

```rust
    let mut led = Pin::new(Port::C, 13, PinMode::Output);

    loop {
        defmt::info!("Program is alive");
        led.set_low(); // when I do this the LED turns on
        delay.delay_ms(1_000_u32);
        led.set_high(); // this line doesn't seem to turn it off
    }
```

is it connected to the timer or am I doing something wrong with the LED itself?

David-OConnor, 2021-08-06:
The light isn't turning off because the code immediately jumps to the top of the loop, setting it low very quickly after setting high. So, you need a second delay after `led.set_high()`. Note that you can use `delay.delay_ms(1_000_);` - no need to specify it's u32.

I'll push a change with updated default speeds for 401 and 411. I'm curious why you're getting the speeds error though.

toudi, 2021-08-06:
ah... this is so obvious now :) everything works perfectly !!

if I can help you in any way I would be more than happy. Also would you want me to commit this tutorial for the blackpill board into the repo? The syntax-overview example is quite large and it took me a while to cut it down to blinky :)

kindest regards

David-OConnor, 2021-08-06:
Go ahead with the example PR - it would be appreciated. In the next version, you will be able to use the default clock config.

To help troubleshoot the `SpeedError`, could you post the output of this?
```rust

let clock_cfg = Clocks {
    plln: 84,
    ..Default::default();
};

defmt::info!("sys: {}, hclk: {}, apb1: {}, apb2: {}", clock_cfg.sysclk(), clock_cfg.hclk(), clock_cfg.apb1(), clock_cfg.apb2());

clock_cfg.setup(&mut dp.RCC, &mut dp.FLASH).unwrap();
```

toudi, 2021-08-06:
here's the output:

F401:
```
 INFO  sys: 84000000, hclk: 84000000, apb1: 21000000, apb2: 21000000
```

F411 (since it's clocked to 100Mhz I changed the line to `plln: 100`):
```
 INFO  sys: 100000000, hclk: 100000000, apb1: 25000000, apb2: 25000000
```

David-OConnor, 2021-08-06:
And it's still giving you the speed error with both those configs? For context, this error is triggered by a validation function that makes sure the various speeds are in limits. At this, I'm not sure which limit is being exceeded. Likely a bug in the validation code, but I don't know where yet.

toudi, 2021-08-07:
hmm sorry I also realized that when I used the other initialization code:

```rust
    let clock_cfg = Clocks {
        input_src: InputSrc::Hsi,
        ..Default::default()
    };
```

then the outpuf on the F401 and F411 is:

```
INFO  sys: 16000000, hclk: 16000000, apb1: 4000000, apb2: 4000000
```

however what I do not understand is that previously this setup code:

```rust
    let clock_cfg = Clocks {
         plln: 84,
         ..Default::default()
    };
```

would simply result in a panic as indicated previously whereas now it works.. no idea, maybe I did `cargo clean` in meantime?

David-OConnor, 2021-08-07:
No idea. Glad both are working. The one where only `plln` is adjusted will be faster: eg 84Mhz for n=84, and 100Mhz for n=100. Using the one that changes input source to HSI results in 16Mhz. (As you inferred from the `sysclk()` printouts we used to troubleshoot.)

Closing this since the issue is solved in a recent commit. Feel fee to continue adding comments etc, or add new issues as you run into other things.

toudi, 2021-08-07:
amazing! thank you so much. So until 0.2.12 is released I need to do the manual setup right? I will definetely prepare a pull request with some tutorial :)

David-OConnor, 2021-08-07:
Yep. The change in 0.2.12 will just alter the defaults, which set a bad config on those 2 variants. The current version will work as long as you ensure the clock config is acceptable for the MCU. (Which simply changing plln does, in this case)

closed 2021-08-10 by David-OConnor