David-OConnor/stm32-hal #21: I2C on STM32L552x panics

Redblueflame, 2021-09-13:
Hi!
I've recently found stm32-hal for my NUCLEO-L552ZE-Q board, and it worked great with the blink example.
I have an OLED display that I wanted to use to try I2C support, but it looks like there's an error in timings calculations, where the program crashes on [i2c.rs:191](https://github.com/David-OConnor/stm32-hal/blob/main/src/i2c.rs#L191), on the fact that scldel is equals to 16 it seems.

Here's the relevant part of the code:
` ` ` rs
#[entry]
fn main() -> ! {
    let cp = cortex_m::Peripherals::take().unwrap();
    let mut dp = pac::Peripherals::take().unwrap();
    clock_cfg.setup(&mut dp.RCC, &mut dp.FLASH).unwrap();
    /* ... */
    // SETUP I2C
    // Configure pins for I2c.
    let mut scl = Pin::new(Port::F, 1, PinMode::Alt(4));
    scl.output_type(OutputType::OpenDrain);

    let mut sda = Pin::new(Port::F, 0, PinMode::Alt(4));
    sda.output_type(OutputType::OpenDrain);

    // Set up an I2C peripheral, running at 100Khz.
    let i2c = I2c::new(dp.I2C2, I2cDevice::Two, 400_000, &clock_cfg); // <- Panics here
    /* ... */
}
` ` `

David-OConnor, 2021-09-13:
I need to go over the I2C code, especially related to clock config - it's likely there's a bug there. I plan to change the API so you set the clock scaler instead of attempting to set a speed. I'll let you know once complete.

What happens if you set 50kHz as the speed instead?

David-OConnor, 2021-09-14:
Redid I2C init and clock code on github. Try now. Note the breaking change. Ie, to set 400K, you now do this:

```
let i2c = I2c::new(
    dp.I2C1, 
    I2cDevice::Two, 
    I2cConfig {
        I2cSpeed::Fast400K,
        ..Default::default()
    },
    &clock_cfg
);
```

Redblueflame, 2021-09-14:
I was able to make it work on 400kHz on my hardware. Thank you very much!

closed 2021-09-14 by Redblueflame