David-OConnor/stm32-hal #22: "No apb2 on G0" when trying to use TIM1

wjbuys, 2021-09-20:
This snippet of code extracted from the syntax example doesn't run on an STM32G070KBT6:

```rust
#[entry]
fn main() -> ! {
    let mut cp = cortex_m::Peripherals::take().unwrap();
    let mut dp = pac::Peripherals::take().unwrap();

    let clock_cfg = Clocks::default();
    clock_cfg.setup(&mut dp.RCC, &mut dp.FLASH).unwrap();

    let mut timer = Timer::new_tim1(dp.TIM1, 0.2, &clock_cfg);

    loop {
        low_power::sleep_now(&mut cp.SCB);
    }
}

```

It produces this panic:
```
0 ERROR panicked at 'not implemented: No apb2 on G0', /home/jacob/.cargo/registry/src/github.com-1ecc6299db9ec823/stm32-hal2-1.1.0/src/clocks/baseline.rs:1231:17
└─ panic_probe::print_defmt::print @ /home/jacob/.cargo/registry/src/github.com-1ecc6299db9ec823/panic-probe-0.2.1/src/lib.rs:95
stack backtrace:
   0: HardFaultTrampoline
      <exception entry>
   1: lib::inline::__udf
        at ./asm/inline.rs:172:5
   2: __udf
        at ./asm/lib.rs:49:17
   3: cortex_m::asm::udf
        at /home/jacob/.cargo/registry/src/github.com-1ecc6299db9ec823/cortex-m-0.7.3/src/asm.rs:43:5
   4: rust_begin_unwind
        at /home/jacob/.cargo/registry/src/github.com-1ecc6299db9ec823/panic-probe-0.2.1/src/lib.rs:76:9
   5: core::panicking::panic_fmt
        at /rustc/a178d0322ce20e33eac124758e837cbd80a6f633/library/core/src/panicking.rs:92:14
   6: stm32_hal2::clocks::baseline::Clocks::apb2_timer
        at /home/jacob/.cargo/registry/src/github.com-1ecc6299db9ec823/stm32-hal2-1.1.0/src/clocks/baseline.rs:1231:17
   7: stm32_hal2::timer::Timer<stm32g0::stm32g070::TIM1>::new_tim1
        at /home/jacob/.cargo/registry/src/github.com-1ecc6299db9ec823/stm32-hal2-1.1.0/src/timer.rs:260:30
   8: demo::__cortex_m_rt_main
   9: main
        at src/bin/demo.rs:32:1
  10: Reset
        at /home/jacob/.cargo/registry/src/github.com-1ecc6299db9ec823/cortex-m-rt-0.6.15/src/lib.rs:526:15
```

Looking at the code in baseline.rs, this looks incorrect:

https://github.com/David-OConnor/stm32-hal/blob/fa02753476f1501779bbbd69307ddd983a2d549c/src/clocks/baseline.rs#L1225-L1233

According to the [STM32G0x0 reference manual](https://www.st.com/resource/en/reference_manual/rm0454-stm32g0x0-advanced-armbased-32bit-mcus-stmicroelectronics.pdf) section 5.4.15 the `ABPENR2` register does exist.

wjbuys, 2021-09-20:
I get a similar error trying to use any peripherals that are in the `APBENR2` register (i.e. USART1, etc). Looks like the macro that defines peripherals assumes that all those peripherals will use the non-existent APB2, but on G0 they use APB1. I'm poking around trying to find all those spots and adding `#![cfg(feature="g0")]` branches, but it feels very hacky...

David-OConnor, 2021-09-20:
Good find! Fixed, I think. If you're using the Github repo as a dep, it should work now.

> Looks like the macro that defines peripherals assumes that all those peripherals will use the non-existent APB2, but on G0 they use APB1

This was the crux of it; just had to set it to use the APB1 prescaler (On Cube, it's evidently just called APB prescaler on G0).

wjbuys, 2021-09-20:
Sweet, confirmed that fixed it. Thanks!

David-OConnor, 2021-09-20:
Just a note - this HAL's API still refers to the relevant prescaler as `apb1` in the `Clocks` struct. This keeps things simple in the code. Just note that on G0, this controls both APB1 and APB2 peripheral clocks.

closed 2021-09-20 by David-OConnor