David-OConnor/stm32-hal #23: Could not link example blink for Nucleo64 STM32F303RE

nils-van-zuijlen, 2021-09-29:
I got the following error trying to compile the blink example for the STM32F303RE.

<details>
<summary>rust-lld error log</summary>

```
error: linking with `rust-lld` failed: exit status: 1
  |
  = note: "rust-lld" "-flavor" "gnu" "/path/to/project/target/thumbv7em-none-eabihf/debug/deps/project-f3b793af3e647795.1zatyt72j2nvv338.rcgu.o" "/path/to/project/target/thumbv7em-none-eabihf/debug/deps/project-f3b793af3e647795.21n7h0o3jctaqwct.rcgu.o" "/path/to/project/target/thumbv7em-none-eabihf/debug/deps/project-f3b793af3e647795.2r5valll98nq0yxt.rcgu.o" "/path/to/project/target/thumbv7em-none-eabihf/debug/deps/project-f3b793af3e647795.308nglvz1fd2kk10.rcgu.o" "/path/to/project/target/thumbv7em-none-eabihf/debug/deps/project-f3b793af3e647795.31gsx5az23k737mo.rcgu.o" "/path/to/project/target/thumbv7em-none-eabihf/debug/deps/project-f3b793af3e647795.3dx73n44l4e015if.rcgu.o" "/path/to/project/target/thumbv7em-none-eabihf/debug/deps/project-f3b793af3e647795.3ntu1gyt5ca3wi38.rcgu.o" "/path/to/project/target/thumbv7em-none-eabihf/debug/deps/project-f3b793af3e647795.44g2xjeux4x46cdg.rcgu.o" "/path/to/project/target/thumbv7em-none-eabihf/debug/deps/project-f3b793af3e647795.4mnhdpm2nyv86la2.rcgu.o" "/path/to/project/target/thumbv7em-none-eabihf/debug/deps/project-f3b793af3e647795.4naaal1ok3c4xemm.rcgu.o" "/path/to/project/target/thumbv7em-none-eabihf/debug/deps/project-f3b793af3e647795.55x5tadwp2rnt0x9.rcgu.o" "/path/to/project/target/thumbv7em-none-eabihf/debug/deps/project-f3b793af3e647795.9b82ovfs2agj1gn.rcgu.o" "/path/to/project/target/thumbv7em-none-eabihf/debug/deps/project-f3b793af3e647795.pazaxvcmqz5u03o.rcgu.o" "/path/to/project/target/thumbv7em-none-eabihf/debug/deps/project-f3b793af3e647795.s9urt31o9qsfti9.rcgu.o" "--as-needed" "-L" "/path/to/project/target/thumbv7em-none-eabihf/debug/deps" "-L" "/path/to/project/target/debug/deps" "-L" "/path/to/project/target/thumbv7em-none-eabihf/debug/build/cortex-m-2fd1a8e6d25874b3/out" "-L" "/path/to/project/target/thumbv7em-none-eabihf/debug/build/cortex-m-rt-2100316a60da11aa/out" "-L" "/path/to/project/target/thumbv7em-none-eabihf/debug/build/defmt-dcf8dbeba0568f57/out" "-L" "/path/to/project/target/thumbv7em-none-eabihf/debug/build/stm32f3-0c9e140b792c9684/out" "-L" "/path/to/project/target/thumbv7em-none-eabihf/debug/build/cortex-m-rt-e97eb40fc8237973/out" "-L" "/home/nils/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/thumbv7em-none-eabihf/lib" "-Bstatic" "/path/to/project/target/thumbv7em-none-eabihf/debug/deps/libdefmt_rtt-0d410999536130db.rlib" "/path/to/project/target/thumbv7em-none-eabihf/debug/deps/libdefmt-f00fe835de2640b0.rlib" "/path/to/project/target/thumbv7em-none-eabihf/debug/deps/libcortex_m-56559b207976fe0d.rlib" "/path/to/project/target/thumbv7em-none-eabihf/debug/deps/libaligned-1a7560077d02064d.rlib" "/path/to/project/target/thumbv7em-none-eabihf/debug/deps/libas_slice-745fcc339b4c8ccb.rlib" "/path/to/project/target/thumbv7em-none-eabihf/debug/deps/libstable_deref_trait-812856d833b7407e.rlib" "/path/to/project/target/thumbv7em-none-eabihf/debug/deps/libgeneric_array-982c30bca4e75e3f.rlib" "/path/to/project/target/thumbv7em-none-eabihf/debug/deps/libgeneric_array-09033a9cd8cd62c4.rlib" "/path/to/project/target/thumbv7em-none-eabihf/debug/deps/libgeneric_array-cf176f5edce2b95e.rlib" "/path/to/project/target/thumbv7em-none-eabihf/debug/deps/libtypenum-ff9467418d54e116.rlib" "/path/to/project/target/thumbv7em-none-eabihf/debug/deps/libstm32_hal2-2c565610c9b9f015.rlib" "/path/to/project/target/thumbv7em-none-eabihf/debug/deps/libchrono-572d54bc91f51079.rlib" "/path/to/project/target/thumbv7em-none-eabihf/debug/deps/libnum_integer-cc30d5d71907a14e.rlib" "/path/to/project/target/thumbv7em-none-eabihf/debug/deps/libnum_traits-25f97b3b71d2574c.rlib" "/path/to/project/target/thumbv7em-none-eabihf/debug/deps/liblibm-cf706319bdd7c8a9.rlib" "/path/to/project/target/thumbv7em-none-eabihf/debug/deps/libcast-0b28ad7b064dfac7.rlib" "/path/to/project/target/thumbv7em-none-eabihf/debug/deps/libcfg_if-f0905b992f2d6a32.rlib" "/path/to/project/target/thumbv7em-none-eabihf/debug/deps/libstm32f3-b1cc477b6ab5f559.rlib" "/path/to/project/target/thumbv7em-none-eabihf/debug/deps/libcortex_m_rt-3aff3035a9133645.rlib" "/path/to/project/target/thumbv7em-none-eabihf/debug/deps/libr0-eb0d4d2f80fb1fe8.rlib" "/path/to/project/target/thumbv7em-none-eabihf/debug/deps/libcortex_m_rt-fc679c9dec16b97a.rlib" "/path/to/project/target/thumbv7em-none-eabihf/debug/deps/libcortex_m-039d81d222b07560.rlib" "/path/to/project/target/thumbv7em-none-eabihf/debug/deps/libembedded_hal-2975e04cc4cac382.rlib" "/path/to/project/target/thumbv7em-none-eabihf/debug/deps/libvoid-47f217d30b4d30b2.rlib" "/path/to/project/target/thumbv7em-none-eabihf/debug/deps/libnb-04defd24f4b073b9.rlib" "/path/to/project/target/thumbv7em-none-eabihf/debug/deps/libnb-0a4ca79a0a5fa29a.rlib" "/path/to/project/target/thumbv7em-none-eabihf/debug/deps/libvolatile_register-8bc92543b32933eb.rlib" "/path/to/project/target/thumbv7em-none-eabihf/debug/deps/libvcell-eb41bddaf9059617.rlib" "/path/to/project/target/thumbv7em-none-eabihf/debug/deps/libbare_metal-fbdd2896b239bab4.rlib" "/home/nils/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/thumbv7em-none-eabihf/lib/librustc_std_workspace_core-f5e50862e11e2490.rlib" "/home/nils/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/thumbv7em-none-eabihf/lib/libcore-87c45edd00288221.rlib" "/home/nils/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/thumbv7em-none-eabihf/lib/libcompiler_builtins-bf4341c12e226c41.rlib" "-Bdynamic" "--eh-frame-hdr" "-znoexecstack" "-L" "/home/nils/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/thumbv7em-none-eabihf/lib" "-o" "/path/to/project/target/thumbv7em-none-eabihf/debug/deps/project-f3b793af3e647795" "--gc-sections" "-Tlink.x" "-Tdefmt.x"
  = note: rust-lld: error: undefined symbol: WWDG
          >>> referenced by stm32f3.c86517d0-cgu.13
          >>>               stm32f3-b1cc477b6ab5f559.stm32f3.c86517d0-cgu.13.rcgu.o:(__INTERRUPTS) in archive /path/to/project/target/thumbv7em-none-eabihf/debug/deps/libstm32f3-b1cc477b6ab5f559.rlib
          
          rust-lld: error: undefined symbol: PVD
          >>> referenced by stm32f3.c86517d0-cgu.13
          >>>               stm32f3-b1cc477b6ab5f559.stm32f3.c86517d0-cgu.13.rcgu.o:(__INTERRUPTS) in archive /path/to/project/target/thumbv7em-none-eabihf/debug/deps/libstm32f3-b1cc477b6ab5f559.rlib
          
          rust-lld: error: undefined symbol: TAMP_STAMP
          >>> referenced by stm32f3.c86517d0-cgu.13
          >>>               stm32f3-b1cc477b6ab5f559.stm32f3.c86517d0-cgu.13.rcgu.o:(__INTERRUPTS) in archive /path/to/project/target/thumbv7em-none-eabihf/debug/deps/libstm32f3-b1cc477b6ab5f559.rlib
          
          rust-lld: error: undefined symbol: RTC_WKUP
          >>> referenced by stm32f3.c86517d0-cgu.13
          >>>               stm32f3-b1cc477b6ab5f559.stm32f3.c86517d0-cgu.13.rcgu.o:(__INTERRUPTS) in archive /path/to/project/target/thumbv7em-none-eabihf/debug/deps/libstm32f3-b1cc477b6ab5f559.rlib
          
          rust-lld: error: undefined symbol: FLASH
          >>> referenced by stm32f3.c86517d0-cgu.13
          >>>               stm32f3-b1cc477b6ab5f559.stm32f3.c86517d0-cgu.13.rcgu.o:(__INTERRUPTS) in archive /path/to/project/target/thumbv7em-none-eabihf/debug/deps/libstm32f3-b1cc477b6ab5f559.rlib
          
          rust-lld: error: undefined symbol: RCC
          >>> referenced by stm32f3.c86517d0-cgu.13
          >>>               stm32f3-b1cc477b6ab5f559.stm32f3.c86517d0-cgu.13.rcgu.o:(__INTERRUPTS) in archive /path/to/project/target/thumbv7em-none-eabihf/debug/deps/libstm32f3-b1cc477b6ab5f559.rlib
          
          rust-lld: error: undefined symbol: EXTI0
          >>> referenced by stm32f3.c86517d0-cgu.13
          >>>               stm32f3-b1cc477b6ab5f559.stm32f3.c86517d0-cgu.13.rcgu.o:(__INTERRUPTS) in archive /path/to/project/target/thumbv7em-none-eabihf/debug/deps/libstm32f3-b1cc477b6ab5f559.rlib
          
          rust-lld: error: undefined symbol: EXTI1
          >>> referenced by stm32f3.c86517d0-cgu.13
          >>>               stm32f3-b1cc477b6ab5f559.stm32f3.c86517d0-cgu.13.rcgu.o:(__INTERRUPTS) in archive /path/to/project/target/thumbv7em-none-eabihf/debug/deps/libstm32f3-b1cc477b6ab5f559.rlib
          
          rust-lld: error: undefined symbol: EXTI2_TSC
          >>> referenced by stm32f3.c86517d0-cgu.13
[... cut ...]
          >>>               stm32f3-b1cc477b6ab5f559.stm32f3.c86517d0-cgu.13.rcgu.o:(__INTERRUPTS) in archive /path/to/project/target/thumbv7em-none-eabihf/debug/deps/libstm32f3-b1cc477b6ab5f559.rlib
          
          rust-lld: error: undefined symbol: DMA1_CH5
[... 31 lines cut ...]

```toml
[package]
name = "menu"
version = "0.1.0"
edition = "2018"

# See more keys and their definitions at https://doc.rust-lang.org/cargo/reference/manifest.html

[dependencies]
cortex-m = "0.7.3"
cortex-m-rt = "0.7.0"
stm32-hal2 = { version = "^1.2", features = ["f303", "f3rt"] }
defmt = "0.2.3"
defmt-rtt = "0.2.0"

[features]
default = [
    "defmt-default",
]

defmt-default = []
```

</details>

David-OConnor, 2021-09-29:
What is the `runner` line set to in `.cargo/config.toml`?

nils-van-zuijlen, 2021-09-29:
`runner = "probe-run --chip STM32F303RETx"`

I use the onboard ST-Link.

nils-van-zuijlen, 2021-09-29:
And that does not seem to matter much, I got the same error when running `cargo build` alone.

David-OConnor, 2021-09-29:
Thank you. Could you please post your dependencies from `Cargo.toml`?

David-OConnor, 2021-09-29:
What happens if you change your `cortex-m-rt` version to `0.6.14`?

nils-van-zuijlen, 2021-09-29:
Downgrading it, running `cargo update`, then `cargo build` compiles.

But now `probe-run` (launched by `cargo run`) fails with the following error:
```
$ cargo run
    Finished dev [unoptimized + debuginfo] target(s) in 0.85s
     Running `probe-run --chip STM32F303RETx target/thumbv7em-none-eabihf/debug/menu`
(HOST) INFO  flashing program (113.96 KiB)
(HOST) INFO  success!
Error: Operation timed out
$
```

nils-van-zuijlen, 2021-09-29:
And trying to flash and run with openocd & gdb, I fall on the breakpoint set on `cortex_m_rt::Reset()` followed by `HardFaultTrampoline()`.

David-OConnor, 2021-09-29:
The `cortex-m-rt` version problem is due to incompatibility between the `stm32-rs` PACs and `cortex-m-rt` 0.7.

What happens if you unplug and plug in the board's USB port? Are you able to erase the flash memory using [STM32CubeProgrammer](https://www.st.com/en/development-tools/stm32cubeprog.html#overview&secondary=st-get-software)? Is `probe-run` up to date?

(I'm not sure what's wrong, but these are all worth trying)

David-OConnor, 2021-09-29:
Another thought: Does your code use `wfi` or `wfe`?

nils-van-zuijlen, 2021-09-29:
> What happens if you unplug and plug in the board's USB port?

Same output

> Are you able to erase the flash memory using STM32CubeProgrammer?

Yes I am able to write all `0x0000_0000` to the whole flash memory.

> Is `probe-run` up to date?

I installed it yesterday, it is.
```
$ cargo install probe-run
    Updating crates.io index
     Ignored package `probe-run v0.2.6` is already installed, use --force to override
$
```

> Does your code use `wfi` or `wfe`?

The only thing I changed from the `blink` example is the led pin. I don't think I'm using those, but as I don't know what they are, I can't be certain.

David-OConnor, 2021-09-29:
What happens if you comment out everything below the `defmt::info!("Hello, world!");` line from the blinky example? (Perhaps the F3 clock default is wrong)

David-OConnor, 2021-09-29:
If that works, try this for clock code:

use stm32_hal2::Clocks::Prediv;

```rust
let clock_cfg = Clocks {
    prediv: Prediv::Div2,
    ..Default::default()
}
clock_cfg.setup().unwrap();
```

David-OConnor, 2021-09-29:
Use this:

```rust
use stm32_hal2::Clocks::PllMul;
// ...

let clock_cfg = Clocks {
    pll_mul: PllMul::Mul8,
    ..Default::default()
}
clock_cfg.setup().unwrap();
```

nils-van-zuijlen, 2021-09-29:
> What happens if you comment out everything below the `defmt::info!("Hello, world!");` line from the blinky example? (Perhaps the F3 clock default is wrong)

I get the same issue, exactly.

Also, when I tried to run it with openocd and gdb, I had a breakpoint on `main` that was never reached, so there's that...

David-OConnor, 2021-09-29:
What does the `memory.x` file in your project root look like?

nils-van-zuijlen, 2021-09-29:
`memory.x`:
```
MEMORY
{
  /* NOTE K = KiBi = 1024 bytes */
  FLASH : ORIGIN = 0x08000000, LENGTH = 512K 
  RAM : ORIGIN = 0x20000000, LENGTH = 80K
}
```

nils-van-zuijlen, 2021-09-29:
Here's the documentation that gave me those addresses (page 60 of the reference manual)
[image]

David-OConnor, 2021-09-29:
Yea - those are correct. I'll let you know what I come up with

David-OConnor, 2021-09-30:
Btw, re the Clocks thing I posted - I looked again at the RM, and the default setting should work. Where you get in trouble is if you try to use anything but the default on F303xD/E variants like this - then it'll be wrong, since those variants have 4 possible PLL source settings, and only the first does the same thing as the other 303 variants!

David-OConnor, 2021-09-30:
Are you able to get [stm32-f3xx-hal](https://github.com/stm32-rs/stm32f3xx-hal) working with your board?

David-OConnor, 2021-11-04:
Closing due to lack of response

closed 2021-11-04 by David-OConnor