David-OConnor/stm32-hal #29: Power not enabled for GPIO on STM32F3xx?

ctamblyn, 2021-11-14:
First of all, thanks for writing this awesome library!

While experimenting with this crate on an STM32F3 Discovery board, it didn't seem like the power was getting enabled for GPIO port E when creating a pin using `Pin::new(...)`.

Here's a fairly minimised example of what I'm doing:

``` rust
#![no_std]
#![no_main]

use panic_semihosting as _;

use cortex_m_rt::entry;
use stm32_hal2::gpio::{Pin, PinMode, Port};

#[entry]
fn main() -> ! {
    let mut led = Pin::new(Port::E, 9, PinMode::Output);
    led.set_high();

    loop {}
}
```

And this is my Cargo.toml:

```
[package]
name = "hello-rtic"
version = "0.1.0"
edition = "2018"

[dependencies]
cortex-m-rt = "0.6.15"
cortex-m-semihosting = "0.3.5"
panic-semihosting = "0.5"

[dependencies.stm32-hal2]
version = "1.3.1"
features = ["f303", "f3rt"]

# this lets you use `cargo fix`!
[[bin]]
name = "hello-rtic"
test = false
bench = false

[profile.dev]
opt-level = 1
codegen-units = 16
debug = true
lto = false

[profile.release]
opt-level = "s"     # optimize for size
codegen-units = 1   # better optimizations
debug = true        # symbols are nice and they don't increase the size on Flash
lto = true          # better optimizations
```

Running the above should cause the LED attached to pin PE9 to light up, but does not.

Poking around in the implementation of `Pin::new()`, I think the cause is the following bit of code:

``` rust
pub fn new(port: Port, pin: u8, mode: PinMode) -> Self {
    // ... snip ...
                Port::E => {
                    cfg_if! {
                        if #[cfg(feature = "f3")] {
                            if rcc.ahbenr.read().iopeen().bit_is_clear() {
                                rcc_en_reset!(ahb1, iopa, rcc);
                            }
    // ... snip ...
}
```

If I'm understanding that correctly, it's writing to IOPAEN (enabling the clock for port A) when it should be writing to IOPEEN (section 9.4.6 of the [reference manual](https://www.st.com/resource/en/reference_manual/dm00043574-stm32f303xb-c-d-e-stm32f303x6-8-stm32f328x8-stm32f358xc-stm32f398xe-advanced-arm-based-mcus-stmicroelectronics.pdf)).  I think ports (other than A) are all similarly affected on the F3.

David-OConnor, 2021-11-15:
Good catch. It was broken for all ports. Fixed in latest commit.

ctamblyn, 2021-11-15:
Can confirm that the latest commit results in a nicely illuminated LED on my board :)

David-OConnor, 2021-11-15:
Sweet!

closed 2021-11-15 by David-OConnor