David-OConnor/stm32-hal #33: IrDA support for USART peripherals

tim-seoss, 2022-03-16:
Thanks for the work on stm32-hal...

I'm working on adding IrDA SIR support for the stm32f4xx-hal currently - would you be interested in adding the same support to this project?  A very-draft PR is here: https://github.com/stm32-rs/stm32f4xx-hal/pull/466 any comments would be welcome.

David-OConnor, 2022-03-16:
Definitely

David-OConnor, 2022-03-17:
In latest commit. I used your cfg enum; works well with the idioms I use in this lib. Untested, and I've never used IrDA. This should probably work (It's straight out of the RM, and consistent with your PR in the other repo), but I didn't understand the part about baud rate cfg you set up.

https://github.com/David-OConnor/stm32-hal/blob/main/src/usart.rs#L232

tim-seoss, 2022-03-18:
Cool, thanks.  If it's OK with you I'll try and get the other PR mostly straightened out first (I've got some doubt about IrDA low power mode on UART peripherals - possibly this functionality is missing for UARTs), and then come back to this?

David-OConnor, 2022-04-24:
Closing for now, since the bulk of the functionality is now included. Feel free to open another issue/PR if you want to add the baud rate cfg.

closed 2022-04-24 by David-OConnor

tim-seoss, 2022-04-25:
Thanks. BTW the stm32f4xx PR referenced above is currently blocked on this peripheral access crate PR: https://github.com/stm32-rs/stm32-rs/issues/711