David-OConnor/stm32-hal #36: U5 Support

gdf8gdn8, 2022-04-09:
Hi,
Please support ST U5 series.
https://www.st.com/en/microcontrollers-microprocessors/stm32u5-series.html#cad-resources

https://www.st.com/resource/en/svd/stm32u5_svd.zip

David-OConnor, 2022-04-09:
Sweet. I guess there's no reason not to now that the SVD's out. I'll have a crack at it when I find time.

David-OConnor, 2022-04-09:
Upon further consideration, I'd like to wait until it's supported in the [stm32-rs repo](https://github.com/stm32-rs/stm32-rs-nightlies), since this library is an abstraction upon the PACs these generate.

potentialdiffer, 2023-02-01:
Hi together, 
just wanted to give a heads up on that the stm32u5 PAC has been merged: https://github.com/stm32-rs/stm32-rs/pull/744

driveraid, 2024-04-23:
Any plans or progress here?

David-OConnor, 2024-04-24:
Hey! I don't have the time ATM to add support for this, but you are welcome if you'd like! It will probably be a bit tedious, but doable.