David-OConnor/stm32-hal #38: [SMT32F446] Compile broken

gfabiano, 2022-04-14:
stm32f4re446 does not compile anymore,

in particular, there are no registers in PAC that and with `_3`.

for example:

```
error[E0599]: no method named `oc1m_3` found for mutable reference `&mut stm32f4::stm32f446::tim1::ccmr1_output::W` in the current scope
    --> stm32-hal/src/timer.rs:1074:31
     |
1074 |                             w.oc1m_3().bit((mode as u8) >> 3!= 0)
     |                               ^^^^^^ help: there is an associated function with a similar name: `oc1m`
```

David-OConnor, 2022-04-14:
Yep. All non-DMAMUX variants (F3, F4, L4) are currently broken on the repo. I made an internal refactor to simplify code across most of the modules regarding DMA channel selection, but it caused this. The most recent release is pre this. I may have to just revert to the old (working, verbose) style.

David-OConnor, 2022-04-14:
*Oh wait actually this is something different I just introduced. Leaving the above comment in since lol since it's also true.

David-OConnor, 2022-04-14:
Fixed

closed 2022-04-14 by David-OConnor