David-OConnor/stm32-hal #41: [SMT32L476] Compile broken

badrbouslikhin, 2022-05-10:
Hi,

STM32L476 does not compile anymore. Here's the ouput when I try to compile [stm32-hal-quickstart](https://github.com/David-OConnor/stm32-hal-quickstart):

```
stm32-hal-quickstart on  main [!] is 📦 v0.1.0 via 🦀 v1.60.0 
❯ cargo run --release 
   Compiling proc-macro2 v1.0.38
   Compiling unicode-xid v0.2.3
   Compiling syn v1.0.93
   Compiling semver-parser v0.7.0
   Compiling version_check v0.9.4
   Compiling cortex-m v0.7.4
   Compiling nb v1.0.0
   Compiling autocfg v1.1.0
   Compiling vcell v0.1.3
   Compiling semver v1.0.9
   Compiling void v1.0.2
   Compiling defmt v0.3.1
   Compiling bitfield v0.13.2
   Compiling libm v0.2.2
   Compiling defmt-macros v0.3.2
   Compiling cortex-m-rt v0.7.1
   Compiling defmt-parser v0.3.1
   Compiling bare-metal v1.0.0
   Compiling bitflags v1.3.2
   Compiling cfg-if v1.0.0
   Compiling stm32l4 v0.14.0
   Compiling critical-section v0.2.7
   Compiling panic-probe v0.3.0
   Compiling defmt-rtt v0.3.2
   Compiling paste v1.0.7
   Compiling nb v0.1.3
   Compiling volatile-register v0.2.1
   Compiling semver v0.9.0
   Compiling proc-macro-error-attr v1.0.4
   Compiling proc-macro-error v1.0.4
   Compiling num-traits v0.2.15
   Compiling num-integer v0.1.45
   Compiling embedded-hal v0.2.7
   Compiling rustc_version v0.2.3
   Compiling bare-metal v0.2.5
   Compiling quote v1.0.18
   Compiling rustc_version v0.4.0
   Compiling cast v0.2.7
   Compiling chrono v0.4.19
   Compiling cortex-m-rt-macros v0.7.0
   Compiling stm32-hal2 v1.4.4
error[E0599]: no method named `pif0` found for mutable reference `&mut stm32l4::stm32l4x6::exti::pr1::W` in the current scope
    --> /Users/badr/.cargo/registry/src/github.com-1ecc6299db9ec823/stm32-hal2-1.4.4/src/gpio.rs:1151:32
     |
1151 |                         0 => w.pif0().set_bit(),
     |                                ^^^^ method not found in `&mut stm32l4::stm32l4x6::exti::pr1::W`

error[E0599]: no method named `pif1` found for mutable reference `&mut stm32l4::stm32l4x6::exti::pr1::W` in the current scope
    --> /Users/badr/.cargo/registry/src/github.com-1ecc6299db9ec823/stm32-hal2-1.4.4/src/gpio.rs:1152:32
     |
1152 |                         1 => w.pif1().set_bit(),
     |                                ^^^^ method not found in `&mut stm32l4::stm32l4x6::exti::pr1::W`

error[E0599]: no method named `pif2` found for mutable reference `&mut stm32l4::stm32l4x6::exti::pr1::W` in the current scope
    --> /Users/badr/.cargo/registry/src/github.com-1ecc6299db9ec823/stm32-hal2-1.4.4/src/gpio.rs:1153:32
     |
1153 |                         2 => w.pif2().set_bit(),
     |                                ^^^^ method not found in `&mut stm32l4::stm32l4x6::exti::pr1::W`

error[E0599]: no method named `pif3` found for mutable reference `&mut stm32l4::stm32l4x6::exti::pr1::W` in the current scope
    --> /Users/badr/.cargo/registry/src/github.com-1ecc6299db9ec823/stm32-hal2-1.4.4/src/gpio.rs:1154:32
     |
1154 |                         3 => w.pif3().set_bit(),
[... 270 lines cut ...]
     |                               ^^^^^^ help: there is an associated function with a similar name: `oc1m_2`
...
1685 | cc_1_channel!(TIM16, u16);
     | ------------------------- in this macro invocation
     |
     = note: this error originates in the macro `cc_1_channel` (in Nightly builds, run with -Z macro-backtrace for more info)

error[E0599]: no method named `oc1m_3` found for mutable reference `&mut stm32l4::stm32l4x6::tim16::ccmr1_output::W` in the current scope
    --> /Users/badr/.cargo/registry/src/github.com-1ecc6299db9ec823/stm32-hal2-1.4.4/src/timer.rs:1273:31
     |
1273 |                             w.oc1m_3().bit((mode as u8) >> 3 != 0);
     |                               ^^^^^^ help: there is an associated function with a similar name: `oc1m_2`
...
1696 |         cc_1_channel!(TIM17, u16);
     |         ------------------------- in this macro invocation
     |
     = note: this error originates in the macro `cc_1_channel` (in Nightly builds, run with -Z macro-backtrace for more info)

For more information about this error, try `rustc --explain E0599`.
error: could not compile `stm32-hal2` due to 34 previous errors
warning: build failed, waiting for other jobs to finish...
error: build failed
```

Thanks for the support!

David-OConnor, 2022-05-10:
Known issue I've been dragging my toes on. Also broken in v1.4.4 release (1.4.3 is fine). Affects all L4 and F3 targets. I did a refactor on the DMA-enable code that appeared to simplify things and reduce repetition, but broke these targets. I'll take a look this week.

badrbouslikhin, 2022-05-10:
Thanks for the feedback!
I tried 1.4.3, 1.4.4, and the git main branch, and they all behave the same.

badrbouslikhin, 2022-05-10:
Probably a completely different issue (do you want me to open a specific issue for this?), but I can't get the STM32 going after flashing:

```
stm32-hal-quickstart on  main [!] is 📦 v0.1.0 via 🦀 v1.60.0 took 7s 
❯ cargo run --release
   Compiling project_name v0.1.0 (/Users/badr/Documents/Personal/bldc/parrot-bldc-controller/rust/stm32-hal-quickstart)
warning: unused imports: `AtomicUsize`, `Ordering`
  --> src/main.rs:11:26
   |
11 | use core::sync::atomic::{AtomicUsize, Ordering};
   |                          ^^^^^^^^^^^  ^^^^^^^^
   |
   = note: `#[warn(unused_imports)]` on by default

warning: unused imports: `Mutex`, `delay::Delay`, `free`, `peripheral::NVIC`
  --> src/main.rs:15:5
   |
15 |     delay::Delay,
   |     ^^^^^^^^^^^^
16 |     interrupt::{free, Mutex},
   |                 ^^^^  ^^^^^
17 |     peripheral::NVIC,
   |     ^^^^^^^^^^^^^^^^

warning: unused imports: `AdcDevice`, `Adc`, `DmaChannel`, `DmaInterrupt`, `Dma`, `Edge`, `OutputType`, `PinMode`, `Pin`, `Port`, `Pull`, `TimerInterrupt`, `Timer`, `self`
  --> src/main.rs:29:11
   |
29 |     adc::{self, Adc, AdcDevice},
   |           ^^^^  ^^^  ^^^^^^^^^
30 |     clocks::Clocks,
31 |     dma::{Dma, DmaChannel, DmaInterrupt},
   |           ^^^  ^^^^^^^^^^  ^^^^^^^^^^^^
32 |     gpio::{Edge, OutputType, Pin, PinMode, Port, Pull},
   |            ^^^^  ^^^^^^^^^^  ^^^  ^^^^^^^  ^^^^  ^^^^
33 |     low_power, pac,
34 |     timer::{Timer, TimerInterrupt},
   |             ^^^^^  ^^^^^^^^^^^^^^

warning: unused variable: `cp`
  --> src/main.rs:40:13
   |
40 |     let mut cp = cortex_m::Peripherals::take().unwrap();
   |             ^^ help: if this is intentional, prefix it with an underscore: `_cp`
   |
   = note: `#[warn(unused_variables)]` on by default

warning: unused variable: `dp`
  --> src/main.rs:42:13
   |
42 |     let mut dp = pac::Peripherals::take().unwrap();
   |             ^^ help: if this is intentional, prefix it with an underscore: `_dp`

warning: variable does not need to be mutable
  --> src/main.rs:40:9
   |
40 |     let mut cp = cortex_m::Peripherals::take().unwrap();
   |         ----^^
   |         |
   |         help: remove this `mut`
   |
   = note: `#[warn(unused_mut)]` on by default

warning: variable does not need to be mutable
  --> src/main.rs:42:9
   |
42 |     let mut dp = pac::Peripherals::take().unwrap();
   |         ----^^
   |         |
   |         help: remove this `mut`

warning: `project_name` (bin "project_name") generated 7 warnings
    Finished release [optimized + debuginfo] target(s) in 1.18s
     Running `probe-run --chip STM32L476RGTx target/thumbv7em-none-eabihf/release/project_name`
(HOST) WARN  (BUG) location info is incomplete; it will be omitted from the output
(HOST) INFO  flashing program (19 pages / 19.00 KiB)
(HOST) INFO  success!
Error: An error with the usage of the probe occured

Caused by:
    Operation timed out
```

Here's the content of `.cargo/config.toml`:
```
runner = "probe-run --chip STM32L476RGTx" # to list chips, run `probe-run --list-chips.`
```

And the content of `memory.x`:
```
/* Change this as required for your MCU */

MEMORY
{
  /* NOTE 1 K = 1 KiBi = 1024 bytes */
  FLASH : ORIGIN = 0x08000000, LENGTH = 1024K
  RAM : ORIGIN = 0x20000000, LENGTH = 128K
}
```

Any ideas?

David-OConnor, 2022-05-10:
Fixed, and published as 1.4.5

For the other thing, make a new issue, and include more details if you have them.

closed 2022-05-10 by David-OConnor

badrbouslikhin, 2022-05-11:
Thanks for your swift support!
The quickstart is also running fine now.