David-OConnor/stm32-hal #42: TIM1/TIM8 PWM mode not working on STM32L476

badrbouslikhin, 2022-05-12:
Hi,

I've been trying to use TIM1 and TIM8 advanced timers on STM32L476 but there seems to be an issue.

The following example doesn't output anything on PC6:
```rust
#![no_main]
#![no_std]

use cortex_m::{self, delay::Delay};
use cortex_m_rt::entry;

// These lines are part of our setup for debug printing.
use defmt_rtt as _;
use panic_probe as _;

// Import parts of this library we use. You could use this style, or perhaps import
// less here.
use stm32_hal2::{
    self,
    clocks::Clocks,
    gpio::{Pin, PinMode, Port},
    pac,
    timer::{OutputCompare, TimChannel, Timer, TimerConfig},
};

#[entry]
fn main() -> ! {
    // Set up ARM Cortex-M peripherals. These are common to many MCUs, including all STM32 ones.
    let cp = cortex_m::Peripherals::take().unwrap();
    // Set up microcontroller peripherals
    let mut dp = pac::Peripherals::take().unwrap();

    stm32_hal2::debug_workaround();

    // Create an initial clock configuration that uses the MCU's internal oscillator (HSI),
    // sets the MCU to its maximum system clock speed.
    let clock_cfg = Clocks::default();

    clock_cfg.setup().unwrap();

    // Set up PWM pin
    defmt::println!("Setting up PWM pins");
    let _pwm_pin = Pin::new(Port::C, 6, PinMode::Alt(3));

    defmt::println!("Configuring PWM timer");
    let mut pwm_timer = Timer::new_tim8(dp.TIM8, 2_400., Default::default(), &clock_cfg);
    pwm_timer.enable_pwm_output(TimChannel::C1, OutputCompare::Pwm1, 0.5);
    pwm_timer.enable();

    // Setup a delay, based on the Cortex-m systick.
    let mut delay = Delay::new(cp.SYST, clock_cfg.systick());

    // Port::A, 5
    let mut led = Pin::new(Port::A, 5, PinMode::Output);

    loop {
        defmt::println!("Looping!");
        led.set_high();
        //delay.delay_ms(100);
        //led.set_low();
        //delay.delay_ms(100);
    }
}

// same panicking *behavior* as `panic-probe` but doesn't print a panic message
// this prevents the panic message being printed *twice* when `defmt::panic` is invoked
#[defmt::panic_handler]
fn panic() -> ! {
    cortex_m::asm::udf()
}

/// Terminates the application and makes `probe-run` exit with exit-code = 0
pub fn exit() -> ! {
    loop {
        cortex_m::asm::bkpt();
    }
}
```

But changing the PWM related lines to use TIM3 on the same PC6 pin works:
```rust
    defmt::println!("Setting up PWM pins"); 
    let _pwm_pin = Pin::new(Port::C, 6, PinMode::Alt(2));

    defmt::println!("Configuring PWM timer");
    // Configure PWM timer
    let mut pwm_timer = Timer::new_tim3(dp.TIM3, 2_400., Default::default(), &clock_cfg);
    pwm_timer.enable_pwm_output(TimChannel::C1, OutputCompare::Pwm1, 0.5);
    pwm_timer.enable();
```

Trying to use TIM1 with [stm32-rs/stm32l4xx-hal](https://github.com/stm32-rs/stm32l4xx-hal) works fine (I couldn't use TIM8, for some reasons it doesn't seem implemented):
```rust
//! Testing PWM output

#![deny(unsafe_code)]
#![deny(warnings)]
#![no_main]
#![no_std]

extern crate panic_halt;

use std::println;

// use cortex_m::asm;
use cortex_m_rt::entry;
use stm32l4xx_hal::{delay, prelude::*, stm32};

#[entry]
fn main() -> ! {
    let c = cortex_m::Peripherals::take().unwrap();
    let p = stm32::Peripherals::take().unwrap();

    let mut flash = p.FLASH.constrain();
    let mut rcc = p.RCC.constrain();
    let mut pwr = p.PWR.constrain(&mut rcc.apb1r1);

    let clocks = rcc.cfgr.freeze(&mut flash.acr, &mut pwr);

    let mut gpioa = p.GPIOA.split(&mut rcc.ahb2);

    // TIM1
    let c1 = gpioa
        .pa8
        .into_alternate(&mut gpioa.moder, &mut gpioa.otyper, &mut gpioa.afrh);

    let mut pwm = p.TIM1.pwm(c1, 1.kHz(), clocks, &mut rcc.apb2);

    let max = pwm.get_max_duty();

    pwm.enable();

    let mut timer = delay::Delay::new(c.SYST, clocks);
    let second: u32 = 100;

    // NB: if the pins are LEDs, brightness is not
    //     linear in duty value.
    loop {
        pwm.set_duty(max);
        timer.delay_ms(second);
        // asm::bkpt();

        pwm.set_duty(max / 11 * 10);
        timer.delay_ms(second);
        // asm::bkpt();

        pwm.set_duty(3 * max / 4);
        timer.delay_ms(second);
        // asm::bkpt();

        pwm.set_duty(max / 2);
        timer.delay_ms(second);
        // asm::bkpt();

        pwm.set_duty(max / 4);
        timer.delay_ms(second);
        // asm::bkpt();
    }
}
```

David-OConnor, 2022-05-12:
I don't have an answer yet, but am suspicious this is related to them being "advanced control timers". I suspect I'm omitting a step. (Have not used them myself)

David-OConnor, 2022-05-12:
Interesting. I suspect then it has to do with how it's being configured for PWM. I also skimmed the L4xx code - of note, they don't impl TIM1 or 8 for normal timers, and for PWM, TIM1 doesn't appear to be diff from the other ones.

David-OConnor, 2022-05-12:
OK, found something relevant in L4xx hal. Try adding this after timer setup, and see if it works. If it does, I'll update this lib. I haven't checked what these do, but noticed L4xx has this for advanced control timers:

```rust
let TIM8 = unsafe { &(*PAC::TIM8::ptr()) };

TIM8.bdtr.write(|w| w.moe().set_bit());
TIM8.egr.write(|w| w.ug().set_bit());
```

David-OConnor, 2022-05-12:
Nice. Fixed on GH. Could you please let me know if that works now without adding that workaround to  your code?

David-OConnor, 2022-05-12:
Nvm, issues with this fix for now; need to figure out the best way to do it

David-OConnor, 2022-05-17:
Added to errata for now; fix would be more macro mess.

closed 2025-08-12 by David-OConnor