David-OConnor/stm32-hal #44: [STM32G0B1] Can't compile quickstart

lmpynix, 2022-06-06:
Attempting to clean build the quickstart repository for STM32G0B1 using the thumbv6m-none-eabi toolchain fails with many errors: [g0_build_errors.txt](https://github.com/David-OConnor/stm32-hal/files/8847801/g0_build_errors.txt).
All I have done is clone the repository, change the toolchain, and alter the feature gates in Cargo.toml to "g0b1" and "g0rt".  Cargo seems to be building the library against stm32g0 v0.14.0 and cortex-m v0.7.5.

David-OConnor, 2022-06-07:
It looks like there's a PAC error on several G0 models including that one, wherein many registers are missing. It was fixed a few months ago: (https://github.com/stm32-rs/stm32-rs/issues/548), but hasn't made it into an `stm32-rs` release yet. So, will need to wait.

David-OConnor, 2022-07-04:
Fixed with PAC update

David-OConnor, 2022-07-04:
Nvm still having a struggle with these variants even after the update. Could probably eliminate certain features or fight through it.

adamt55, 2022-11-14:
Is there any way I could provide some help here? I tried compiling the stm32g0-crate with the new SVDs from STM released in march, which compiled fine as far as I could tell when I swapped out the SVD for the G0B1. I then tried to link it to the HAL here, but this resulted in the attached buildlog. New to embedded, so any pointers appreciated.
[build_g0b1.txt](https://github.com/David-OConnor/stm32-hal/files/9998822/build_g0b1.txt)

David-OConnor, 2022-11-14:
That's an easy fix on the HAL side. Before proceeding with that fix, we'd need to get your new SVD build in a stm32-rs PAC release. Any luck convincing those guys to merge/release?

If you'd like to use a fork of this HAL with your custom built PAC and a path or git dep on it, you'd just need to modify the related feature gate code to excempt the relevant G0 variants from the line that uses DMA as DMA1.

adamt55, 2022-11-16:
Thanks for the input. I got the SVD patch in the stm32-rs repo and they build their nightly-repo accordingly (no new release yet though). What do you mean by excempting my variant from the DMA as DMA1 line? I wrote a feature gate that disabled those `use` statements, but I am still getting the same errors. Could you provide me an example/some documentation I could look up?

adamt55, 2022-11-20:
Alright, I tried fixing some stuff on the PAC as well as some stuff on the HAL. I am down to these errors, which I cannot really wrap my head around. It seems something with the DMA is screwed, or maybe I named the fields wrong somehow? Do you have some insights on that maybe?
[build.txt](https://github.com/David-OConnor/stm32-hal/files/10048316/build.txt)

I've opened a PR on the PAC [here](https://github.com/stm32-rs/stm32-rs/pull/785). I'll open a PR on the changes I've made and maybe you could point me to something there.

David-OConnor, 2022-11-20:
Nice work!

Re the DMA stuff: It's likely that the PAC is set up differently for these variants, for one reason or another. For example, this error:
```rust

error[E0615]: attempted to take value of method `ch1` on type `D`
   --> src/dma.rs:692:46
    |
692 |                         let ccr = &self.regs.ch1.cr;
    |                                              ^^^ method, not a field
    ```
    
    Current code:
```rust
                cfg_if! {
                    if #[cfg(any(feature = "f3", feature = "g0"))] {
                        let ccr = &self.regs.ch1.cr;
                    } else {
                        let ccr = &self.regs.ccr1;
                    }
                }
```

Future code:
```rust
                cfg_if! {
                    if #[cfg(any(feature = "f3", feature = "g071, g081 (etc)"))] {
                        let ccr = &self.regs.ch1.cr;
                    } else if #[cfg(feature = "g0b1")] {
                        let ccr = &self.regs.ccr1();
                    else {
                        let ccr = &self.regs.ccr1;
                    }
                }
```

Of note, the DMA module for variants other than H7 is a bit of a mess due to the way the channels are meched in the PAC. The H7 PAC has nice array fields though.

David-OConnor, 2025-08-07:
Try now. Just fixed and released. B1 and C1 should now work. B0 still broken. (Something minor in the timer module I think)

CI now checks these variants for compile errors.

closed 2025-08-07 by David-OConnor