David-OConnor/stm32-hal #45: GPIO_ASCR not used when setting ADC

AShriki, 2022-06-17:
I noticed that the required GPIO_ASCR register (connects gpio to analog mux) for the l4x6 family is not used. I fixed it locally for L4x6by changing 
https://github.com/David-OConnor/stm32-hal/blob/main/src/gpio.rs#L52
to
`#[derive(Copy, Clone, PartialEq)]`

David-OConnor, 2022-06-17:
Done, on repo. Instead of automatically setting as analog, added a l4x6-only method: `connect_to_adc()`.

And... interesting! The feature-gating here is a bit more subtle than it seems: The issue lies in the PAC breaking out the L4 series as `l4x6` etc, but per the RM:

> For the ADC, DAC, OPAMP and COMP, configure the desired I/O in analog mode
in the GPIOx_MODER register and configure the required function in the ADC,
DAC, OPAMP, and COMP registers. For the ADC, it is necessary to configure the
GPIOx_ASCR register (only for STM32L47x/L48x)

So, this is only on L47x and L48x, and only when using with the ADC. (The register def also confirms it's an ADC-only thing). From a skim, it seems that the main (only?) 47x and 48x variants are 476, and 486, so I'm happy with the way you set this up! AFAIK, this functionality isn't included any other STM32 families; only those L4 variants.

closed 2022-06-17 by David-OConnor

AShriki, 2022-06-17:
Thanks! If you think that's interesting the L1 series takes a similar, but slightly different approach to adc muxing, but I didn't read into it too much.

David-OConnor, 2022-06-18:
In new release