David-OConnor/stm32-hal #46: `rcc_en_reset` doesn't work when `paste` is not imported

kellda, 2022-07-05:
When using the `rcc_en_reset` macro, I get errors that `paste` and `cfg-if` are not found. Adding them to `Cargo.toml` fixes the errors, but it should not be needed.

A solution would be to re-export the `paste` and `cfg_if` macros (maybe with `#[doc(hidden)]`) and use `::stm32_hal2::paste` and `::stm32_hal2::cfg_if` instead.

David-OConnor, 2022-07-05:
Could you give an example of context? `rcc_en_reset` is intended for internal use only.

kellda, 2022-07-05:
So maybe I didn't understood how to use it. I thought I had to enable each peripheral with this macro before to use it.

David-OConnor, 2022-07-05:
Nope - that's handled in the constructors, which call `rcc_en_reset`. I think this confusion was caused by Rust's non-standard export mechanics of macros compared to normal fns and structs.

David-OConnor, 2022-07-05:
FWIW, I tried removing the `#[macro_export]` line above that macro so it's not exposed publicly, but it stopped working privately when I did.

David-OConnor, 2022-07-05:
I'm reclassifying this as a [fixed] bug. Switched to a diff syntax instead of `#[macro_export]`, and it's now `pub(crate)` visibility.