David-OConnor/stm32-hal #47: embedded-hal feature broken in SPI and USART files on L476

AShriki, 2022-07-05:
The errors follow (caused by crate nb being private in embedded_hal)
```
   Compiling stm32l4 v0.15.1
   Compiling stm32-hal2 v1.5.0 (/home/___/projects/sandbox/stm32-hal)
error[E0432]: unresolved import `nb`
  --> stm32-hal/src/usart.rs:40:5
   |
40 | use nb;
   |     ^^ no external crate `nb`

error[E0433]: failed to resolve: use of undeclared crate or module `nb`
   --> stm32-hal/src/spi.rs:891:27
    |
891 |     fn read(&mut self) -> nb::Result<u8, Error> {
    |                           ^^ use of undeclared crate or module `nb`

error[E0433]: failed to resolve: use of undeclared crate or module `nb`
   --> stm32-hal/src/spi.rs:898:37
    |
898 |     fn send(&mut self, byte: u8) -> nb::Result<(), Error> {
    |                                     ^^ use of undeclared crate or module `nb`

error[E0433]: failed to resolve: use of undeclared crate or module `nb`
   --> stm32-hal/src/spi.rs:894:31
    |
894 |             Err(e) => Err(nb::Error::Other(e)),
    |                               ^^^^^ not found in `nb`
    |
help: consider importing one of these items
    |
5   | use cast::Error;
    |
5   | use core::fmt::Error;
    |
5   | use crate::flash::Error;
    |
5   | use crate::i2c::Error;
    |
      and 3 other candidates
help: if you import `Error`, refer to it directly
    |
894 -             Err(e) => Err(nb::Error::Other(e)),
894 +             Err(e) => Err(Error::Other(e)),
    |

error[E0433]: failed to resolve: use of undeclared crate or module `nb`
   --> stm32-hal/src/spi.rs:901:31
    |
901 |             Err(e) => Err(nb::Error::Other(e)),
    |                               ^^^^^ not found in `nb`
    |
help: consider importing one of these items
    |
5   | use cast::Error;
    |
5   | use core::fmt::Error;
    |
5   | use crate::flash::Error;
    |
5   | use crate::i2c::Error;
    |
      and 3 other candidates
help: if you import `Error`, refer to it directly
    |
901 -             Err(e) => Err(nb::Error::Other(e)),
901 +             Err(e) => Err(Error::Other(e)),
    |
```

David-OConnor, 2022-07-06:
What features are you running? `nb` is imported internally in this lib if the `embedded_hal` is selected. For example, it compiles for me with:

```rust
cargo b --features "l4x2 embedded_hal"
``

AShriki, 2022-07-06:
It looks like the feature name was the issue. Thanks.

closed 2022-07-06 by AShriki