David-OConnor/stm32-hal #48: f303 stuck at pplstart

MauroMombelli, 2022-07-09:
tested with both external and default config, it get stuck at 
while rcc.cr.read().pllrdy().is_ready() {}, line 488 of f.rs

David-OConnor, 2022-07-10:
Sounds like a bug! Does it work if you configure the clock source to be HSI or HSE? (Without PLL)

MauroMombelli, 2022-07-10:
I just tried to replicate from the quickstart project, as playing around with linker flags and similar i managed to make it work, but somehow #[defmt::panic_handler] wont work anymore and i need to use a standard #[panic_handler]

Clean quickstart will hardfault on startup

only changes are: stm32-hal2 = { version = "^1.4.5", features = ["f303", "f3rt"]}
and
runner = "probe-run --chip STM32F303VCTx" # to list chips, run `probe-run --list-chips.`
(flash and ram should be ok)

from probe-rs:
(HOST) INFO  flashing program (17 pages / 17.00 KiB)
(HOST) INFO  success!
Error: An error with the usage of the probe occured

from openocd
[image]

interestingly the main() is never hit, and all exception frame seems 0; the old project now work tho. not really sure what is going on here.

David-OConnor, 2022-07-10:
Have you adjusted `memory.x`?

MauroMombelli, 2022-07-10:
yes, just double checked both flash and ram, also verified with the working project

David-OConnor, 2022-07-10:
Are you using a current version of probe-run ?

MauroMombelli, 2022-07-11:
yes, using both probe-run latest version (installed yesterday) and release mode, simply run with `cargo run`.
Also tested with openocd + gdb (setup work with the other project, so i am pretty confident)

David-OConnor, 2022-08-13:
What's the word?

lestofante, 2022-08-13:
what is the meaning of "What's the word?"?

David-OConnor, 2022-08-13:
Status

David-OConnor, 2022-08-18:
Closing due to lack of input.

closed 2022-08-18 by David-OConnor

MauroMombelli, 2022-10-15:
tested against current version, 1.5.3, the issue seems to be fixed