David-OConnor/stm32-hal #56: Filtered IMU example code discrepancies(?)

762SPR, 2022-10-05:
Hello, Thank you for this great HAL! I find it much more intuitive than the embedded-HAL based HALs.
I am writing a program for the L432 that uses RTIC, SPI, and DMA to read from a sensor so I have been referencing your filtered IMU example and have come across a couple places that the code needs to be tweaked to compile. Not sure if this is actually an issue or if it is more of a versioning or issue specific to the L4 chips, but I figured I would bring it up just in case!

First is with the clock config:
https://github.com/David-OConnor/stm32-hal/blob/9f57164b48bd77e2a0c7894f1ae886434d462ba6/examples/spi_imu_filtered.rs#L370
I have found this needs to be:
`let clock_cfg = Clocks::default();`

Next, clearing the DMA interrupt:
https://github.com/David-OConnor/stm32-hal/blob/9f57164b48bd77e2a0c7894f1ae886434d462ba6/examples/spi_imu_filtered.rs#L439
Needs a channel input. I think this may be an L4 thing, since they handle channels/streams a little different...
`dma.clear_interrupt(DmaChannel::C2, DmaInterrupt::TransferComplete);`

And finally, the DMA import seems to be missing. Easy fix with rust analyzer but just letting you know :)

https://github.com/David-OConnor/stm32-hal/blob/9f57164b48bd77e2a0c7894f1ae886434d462ba6/examples/spi_imu_filtered.rs#L41
`spi::Spi, dma::{Dma, DmaChannel, DmaInterrupt, ChannelCfg, Priority, Circular, IncrMode}`
(I have some extras in here because I am setting up configurations for each channel vs. using the defaults.

Again, thank you so much for the work you have put into this!

David-OConnor, 2022-10-06:
Thanks for finding these.

Re clock cfg: Calling `Default::default()` should be identical to
`Clocks::default()`; what error are you getting?

Clearing DMA interrupt: Fixed

DMA import: Fixed.

Thx again!

762SPR, 2022-10-06:
I forget the initial error when I tried it, I just put a note in the code that it had to be changed. However, I just new tried to change it back and it gave me an error on the next line at
`clock_cfg.setup().unwrap();`
That: "type annotations needed type must be known at this point"

So that may have been why I changed it. Of course forcing the type with
`let clock_cfg: Clocks = Default::default();`
made that go away. Perhaps it is just an IDE discrepancy, I am using VSCodium with Rust Analyzer, maybe your setup can figure it out easier than RA? 

All these are pretty trivial anyway and probably addressed by more experienced users without a second thought but as a beginner it took me a little bit to work through them. Hopefully the next newbie will have a better time of it!

David-OConnor, 2022-10-11:
Closing as completed. LMK if you learn any more on the Default:: vs Clocks::

closed 2022-10-11 by David-OConnor