David-OConnor/stm32-hal #59: Could you provide an example to implement rtic_monotonic::Monotonic trait?

seanybaggins, 2022-10-13:
It would be nice if there was an example to implement the Monotonic trait using the hal api calls.

Here is an example using the peripheral access crate. 
https://github.com/kalkyl/f411-rtic/blob/c1c74d887e3620bddcaf9217c93767e9f743d7b1/src/mono.rs

David-OConnor, 2022-10-14:
Good idea. I personally haven't used Monotomic (have been using RTIC just for the interrupt and shared-state functionality), but agree that would be a good addition.

David-OConnor, 2022-10-30:
In the latest release. I think there's a bug with it; can't get it to work properly. Hopefully an easy fix; LMK if you have any ideas, or if you're able to get it working as-is.

David-OConnor, 2022-10-31:
Nice! Check the current impl here: https://github.com/David-OConnor/stm32-hal/blob/main/src/timer.rs#L791

Overall, same idea as what you posted, with the following exceptions:
- Uses `core::Duration`, and a custom `Instant` type (see the newly-added `instant` module in this lib)
- Delegates `now` to a non-trait-requiring `time_elapsed` fn: I think this function will be useful even without using `Monotonic`, for getting the time in seconds, ms, us, or ns since timer start.
- Doesn't work when I try to spawn a Monotonic task in RTIC. Any ideas on that?

seanybaggins, 2022-10-31:
Are you calling `spawn_after` or `spawn`. It could be the timer is hitting the auto_reload value  and resetting before it hits the value you asked to wait for.

David-OConnor, 2022-10-31:
Looks like `spawn` works; tried increasing the timer timeout period, but something is still not working.

David-OConnor, 2023-02-24:
I'm dropping further integration due to the direction RTIC is moving.

closed 2023-02-24 by David-OConnor

seanybaggins, 2023-02-24:
Can you elaborate on `due to the direction RTIC is moving`?

David-OConnor, 2023-02-24:
Async for RTIC 2.

seanybaggins, 2023-02-24:
I wonder if it will play nice with embassy. It would be cool if you could use an RTIC 2 executor with embassy hals. Never used embassy's hals but have wanted to give them a try.

David-OConnor, 2023-02-24:
From what I gather from some chats, it should play nice with Embassy, although I don't know how the interactions will work. I think the intent is to keep the core RTIC functionality (Resource sharing and interrupts) functionality the same, but to move everything else, which would include monotonic, to async.