David-OConnor/stm32-hal #61: Breaking changes from stm32-rs nightlies to be expected

FerdinandvHagen, 2022-10-17:
Current stm32-rs nightlies introduce breaking changes (especially to DFSDM).
Before, each register in a cluster was a dedicated field, now it's encapsulated into arrays.
See also this issue on svd2rust: https://github.com/rust-embedded/svd2rust/issues/578

This obviously breaks the crate:

Example file:
https://github.com/stm32-rs/stm32-rs-nightlies/blob/master/stm32h7/src/stm32h735/dfsdm.rs

Example error (there are about 67 in the dfsdm file which are all similar):
```rust
error[E0609]: no field `flt2` on type `R`
    --> src/dfsdm.rs:1392:46
     |
201  | impl<R> Dfsdm<R>
     |      - type parameter 'R' declared here
...
1392 |                         let cr2 = &self.regs.flt2.cr2;
```

As a fix those parameters will need to be changed to use array indexing instead.

David-OConnor, 2022-10-17:
Thanks for the heads-up. This will take some work, but ultimately this is a good change; will make that module (and any other it affects) much more concise.

Will make the change after the PACs are updated on crates.io.

FerdinandvHagen, 2022-10-17:
As this is a change in the upstream svd2rust crate (so in the generator and not in how the patches are done to the svds in stm32-rs) I'm expecting this to happen across all variants that use clustering for registers (not all do).