David-OConnor/stm32-hal #63: probe-run timeout error

kacper-cholewinski, 2022-10-22:
Hi,
I try to flash quickstart code onto my NUCLEO - STM32F303K8T6, but it goes like this:
`(HOST) INFO  flashing program (18 pages / 18.00 KiB)
(HOST) INFO  success!
Error: An error with the usage of the probe occurred

Caused by:
    Operation timed out
error: process didn't exit successfully: `probe-run --chip STM32F303K8Tx target\thumbv7em-none-eabihf\release\project_name` (exit code: 1)`

I've updated .cargo/config.toml and memory.x, and generally did all things according to instructions.

Solutions will be appreciated :)

David-OConnor, 2022-10-22:
Are you able to connect using Stm32CubeProgrammer?

kacper-cholewinski, 2022-10-22:
Hi, thanks for quick answer.

Yes, I can connect and flash program using Cube without problems.

Moreover, I've tried to flash old embassy-rs project to same board, using probe-run and it didn't work either. Few months ago I did it without any problems.

David-OConnor, 2022-10-27:
Weird! No idea. Is `probe-run` up to date? ie `cargo update prob-run --force`.

kacper-cholewinski, 2022-10-28:
Unfortunately, it is. Is there a way to check what is exactly wrong with the flash? Or maybe I need to post issue at probe-run repo?

kacper-cholewinski, 2022-11-01:
Ok, so I used verbose flag on probe-run and I got this:
`couldn't find valid stack range, not placing stack canary`
What does it mean?

kacper-cholewinski, 2022-11-01:
Ok, but now I've updated hal to latest version and flashing works fine.

I assume that something changed in this release, or maybe rust compiler just hates me. :)

closed 2022-11-01 by kacper-cholewinski

kacper-cholewinski, 2022-11-01:
I've tested some more and my conclusion is - I made mistake in memory.x file at first and corrected it later, but there is need for cargo clean for changes in memory.x to take effect.

I hope at least it will help someone else.