David-OConnor/stm32-hal #67: Broken doc.rs

lestofante, 2022-11-18:
I have a strange issue with the docs at https://docs.rs/stm32-hal2/latest/stm32_hal2/spi/struct.SpiConfig.html
if i click "Mode" 
[image]

i land to a "no result found" page. 

not sure what is going on there, if a bug in the doc.rs or this project

David-OConnor, 2022-11-18:
Oh weird! Same. I wonder if it has to do with the `embedded_hal` feature gate on the `SpiModeType` that it should link to. It looks like `SpiMode` is missing entirely from the docs.

David-OConnor, 2023-01-17:
If anyone has thoughts or fixes on this, LMK

David-OConnor, 2025-05-16:
Should be good now; no longer building docs with EH.

closed 2025-05-16 by David-OConnor