David-OConnor/stm32-hal #69: Question : Hal to use RF on STM32WB55

yann77500, 2023-01-04:
Hello
I am starting a new project, I investigate to find the most adequat Hal crate for the MCU (STM32WB55).  I would to start an application that send data via USB on a first time, and send data by BLE in a second time thanks to stm32wb55 MCU. I am using the STM32WB55 Nucleo board. 

Unfortunatly I am a litlle bit frustrated because hals seems be incomplete or incompatible each together. The hal stm32-hal2 seems to me written pretty good, with examples and the repository is very often updated, but this hal doesn’t contains any RF part of stm32WB55. I follow the hal advised for RF usage « stm32wb_hal » but some examples for periph usages are out of dated, so it is a little bit difficult to me to use this crate.

It’s not a good idea to try to use the two hal in the same time because type mismatch or borroing data like RCC are difficult to manage. 

Do you have any advise ? Rust hal are not mature enought for the mcu stm32wb55 ? 
Thank you by advance for yours answsers

David-OConnor, 2023-01-04:
Hi! Unfortunately, I think I'm providing the answer you don't want to hear: This library has no RF functionality. :( I understand that this places a severe limit on its utility on that MCU. It will work for most of the non-RF functionality on your MCU. I suspect you're correct that there are no mature rust HALs that handle the WB's RF functionality. In general, Rust has poor support for RF.

I'm not sure what the best path is: Perhaps the Rubble library can help? Perhaps this HAL (or the other one you looked at) could be patched to better use the hardware semaphores ST provides to communicate between the cores? I believe you need to flash proprietary ST firmware to the RF core. Then you would (somehow?) communicate between the main core running Rust with the other core; I'm not sure how though.

David-OConnor, 2023-01-05:
Update: Apparently I tried writing HSEM and IPCC modules at some point: https://github.com/David-OConnor/stm32-hal/blob/main/src/hsem.rs https://github.com/David-OConnor/stm32-hal/blob/main/src/ipcc.rs

These are the hardware periphs for communicating between cores. They are incomplete and untested.

yann77500, 2023-01-05:
Hi David, 
Thank you for your quick reply. 
Yes I haven’t notice that your crate have already ipcc and mailbox implementation (I was looking for in examples when I search how implement a new peripheral)
Your peripherals seems be interesting, may be it could be possible to integrate the some content of RF functionality from repo stm32wb55-hal (https://github.com/eupn/stm32wb-hal) into this crate if necessary. I don’t know RF stacks, on BLE side I only want/need an API to open an serial communication over BLE.  
I was asking which hal use to begin project to avoid to change hal in middle of developpement, that is very auwfull. 
First, I will start my project with your hal, I will see how to integrate RF BLE functionnality , if I am able to do it.

Thank you for your time

hgaiser, 2025-10-20:
@David-OConnor I wonder if anything changed in the past years when it comes to RF and HAL support in Rust. I'm looking into using a [NUCLEO-WL33CC2](https://www.st.com/en/evaluation-tools/nucleo-wl33cc2.html) for some RF application, I was hoping I could use Rust.

Would love to hear your thoughts 👍

David-OConnor, 2025-10-20:
The traditional WL's (e.g. STM32WLE5) radio is supported via a separate crate (`semtech_radios`). That one isn't; I hadn't heard of it before. Looks like it uses a new radio developed by ST; cool!

If you're interested, you would need to dig through the Reference manual, and PR a new module for it.