David-OConnor/stm32-hal #76: Question: SPI always reading 255

Simonrazer, 2023-02-21:
Heyo, I have another question. I have a STM32F407G-DISC1. I want to use the accelerometer on it with SPI.
I read a bunch of threads/the datasheets of the board, but all I read over SPI is 255, no matter what register.
Based on this table im pretty sure I am using the right pins.
[image]
[image]

Here is (the interesting part of) my code:   
       
        //Pins
        let mut sck = Pin::new(Port::A, 5, PinMode::Alt(5));
        let mut miso = Pin::new(Port::A, 6, PinMode::Alt(5));
        let mut mosi = Pin::new(Port::A, 7, PinMode::Alt(5));
        let mut cs = Pin::new(Port::E, 3, PinMode::Output); 

        //Set SPI mode 0, rest default
        let spi_cfg = SpiConfig{mode: SpiMode::mode0(), ..Default::default()};
        let mut spi = Spi::new(dp.SPI1, spi_cfg, BaudRate::Div32); //Mode and Baud Rate are correct, working C++ code uses the same

        cs.set_low(); //Chip Select for SPI/the Accelerometer is active low
        let write_buf1 = [0x20, 0b01000111];//The Acc. is of by default, enable it by writing those regss
        spi.write(&write_buf1).ok();
        cs.set_high();

        cs.set_low();
        let mut rb:[u8;1] = [0x20]; //The register is r/w so lets see if that write did anything
        spi.transfer(&mut rb);
        println!("{:b}", rb); //prints: 11111111
        //No matter where I read, i get 255..

I know this isnt really a place to ask this stuff, but I am at a total loss. Any help is greatly appreciated!!

David-OConnor, 2023-02-21:
What happens if you set CS high after the transfer, and before printing the buf?

Simonrazer, 2023-02-21:
No change, 255 all the way

David-OConnor, 2023-02-21:
Are you able to read the device id? (SHould be a register for that).

Simonrazer, 2023-02-21:
Ah okay, that makes sense.
Unfortunatly it didnt help. Still having 255 in all places of the reading buffer when printing.
Yes the ID register (WHO_AM_I i guess) is also 255, even tho it shouldn't be

David-OConnor, 2023-02-21:
What accel/IMU is it?

Of note, AFAIK other than the things I mentioned, your code looks right. (Setting CS high after an operation is required on some SPI devices, but not others). Of note, I'm using an SPI Acc/Gyro on one of my projects, although that's on H7 and G4. It could be a bug in this lib's F4 SPI implementation - I haven't tested that.

David-OConnor, 2023-02-21:
From that it sounds like the clocks are fine.

Simonrazer, 2023-02-22:
Unforunatly I fail to get the other HAL to work enough. I cant get rprintl to work, or actually anything else than blinking LEDs 😅(Really appreciate the much better docs+examples here! ^^)

Another thing, the F407 has 3 SPI devices. When using the second one all communication of USB just fails until I erase the flash. The third one only prints 0 so far, but as it doesnt share pins with the Motion sensor I think that may even make sense for once. Going to find cables to connect them today

David-OConnor, 2024-02-22:
Any updates on this?

Simonrazer, 2024-02-27:
Sorry, not really. I switched to I2C and an external IMU, that worked. Code is here: https://github.com/Simonrazer/Embedded-Rust-Demo/blob/main/src/main.rs

closed 2024-02-27 by David-OConnor