David-OConnor/stm32-hal #79: Unable to toggle GPIO PG12 on stm32l476 eval board

seanybaggins, 2023-03-04:
Currently unable to toggle the pin for the stm32 PG12 pin. My guess is power is not being enabled on port G for the board.

David-OConnor, 2023-03-04:
Good catch. You are correct. Code was enabling port H instead of G. Fixed (on GH).

seanybaggins, 2023-03-06:
Can we reopen this I am still unable to toggle this gpio. Looking at these directions...
[image]

If I try to follow step 1 & 2 using the following code
```rust
                        let pwr = unsafe { &(*crate::pac::PWR::ptr()) };
                        pwr.cr2.modify(|_, w| w.pvme2().set_bit());
                        while pwr.sr2.read().pvmo2().bit_is_set() {}
                        pwr.cr2.modify(|_, w| w.pvme2().clear_bit());
                        pwr.cr2.modify(|_, w| w.iosv().set_bit());

```

I hang on the while loop.

If I try to just set the iosv bit using
```rust
 pwr.cr2.modify(|_, w| w.iosv().set_bit());
```
the bit does not get set.

Any ideas?

David-OConnor, 2023-03-06:
Strange! Are you able to get it working using the PAC directly?

seanybaggins, 2023-03-06:
No. I get the same result. Currently checking if this is a hardware configuration error.

seanybaggins, 2023-03-06:
I got this working in the stm32l4xx-hal by modifying the blinky example. Looks to be a software problem.

David-OConnor, 2023-03-07:
Is the port G RCC getting enabled in the RCC, as read using the PAC?

seanybaggins, 2023-03-08:
Yes.
Working stm32l4xx-hal
```
GPIOG Start
Register 0x48001800: 0xFDFFFFFF

AHB2 ENR Start
Register 0x4002104C: 0x00000040

AHB2 RSTR Start
Register 0x4002102C: 0x00000000

IOSV
Register 0x40007004: 0x00000200
```

stm32hal2
```
GPIOG Start
Register 48001800: 0xFDFFFFFF

AHB2 ENR Start
Register 4002104c: 0x00000044

AHB2 RSTR Start
Register 4002102c: 0x00000000

IOSV
Register 40007004: 0x00000000

```

David-OConnor, 2023-03-08:
I've added the IOSV enable to GPIOG port enable on those L4 variants. I don't expect this to fix the problem because you already tried enabling it manually. But, it's a step in the right direction, since as you point out, this is a mandatory step. Might be worth trying again though with the update. (And checking, after you set up a Pin on Port G, that if IOSV stays enabled.

seanybaggins, 2023-03-08:
Same problem. Thanks for adding that change though. I am currently trying to find what other registers need to be set so the iosv bit can be enabled.

seanybaggins, 2023-03-08:
Okay figured it out.
I had to first enable power to the PWR peripheral. Then I was able to modify the iosv bit.

This is how I am using the code at its current revision. It would be nice If we could add the pwren() call inside the hal.
```rust
        let rcc = unsafe { &(*pac::RCC::ptr()) };
        rcc.apb1enr1.modify(|_, w| w.pwren().set_bit());
        let cs = Pin::new(Port::G, 12, PinMode::Output);
```

seanybaggins, 2023-03-08:
Same thing holds true for USB initialization. Need to enabled power, then change the usv bit. Note that changing the usv or iosv bit required me to re-enable power to PWR between bit toggles of usv and iosv.

seanybaggins, 2023-03-08:
To be more explicate...
```rust
        // Enable Port G Power
        let rcc = unsafe { &(*pac::RCC::ptr()) };
        rcc.apb1enr1.modify(|_, w| w.pwren().set_bit());

        let cs = Pin::new(Port::G, 12, PinMode::Output);

        // Enable Usb Power
        rcc.apb1enr1.modify(|_, w| w.pwren().set_bit()); // <- Was not able to ommit this line even though I already call pwren().
        let pwr = unsafe { &*PWR::ptr() };
        pwr.cr2.modify(|_, w| w.usv().set_bit());
```

seanybaggins, 2023-03-08:
I'm going to make changes to the call and submit a PR if everything works.

seanybaggins, 2023-03-08:
Can now use the above example in the following way...
```rust
        // Note: No more low level register configurations to the user of the hal.
        let cs = Pin::new(Port::G, 12, PinMode::Output);

        // Note: User is still on the hook to properly configure pwr when working with USB. This could definitely be a future improvement when working with USB.
        // Enable Usb Power
        rcc.apb1enr1.modify(|_, w| w.pwren().set_bit()); // <- Was not able to ommit this line even though I already call pwren().set_bit().
        let pwr = unsafe { &*PWR::ptr() };
        pwr.cr2.modify(|_, w| w.usv().set_bit());
```

David-OConnor, 2023-03-09:
Great stuff!

closed 2023-03-09 by David-OConnor