David-OConnor/stm32-hal #88: STM32H503 support?

astapleton, 2023-07-21:
Hi there! I'm looking at starting work on an H503 in near future. I'm considering using this crate, but I see a number of H5 lines [commented out](https://github.com/David-OConnor/stm32-hal/blob/main/Cargo.toml#L145-L148) in the Cargo.toml file. What is the state of support? Is it blocked by anything? Does it require contributions?

David-OConnor, 2023-07-22:
Hi. It's untested, with a mix of peripheral support. So, you will almost surely find bugs that may or may not be easy to patch.

David-OConnor, 2023-07-22:
*Ohhh. The piece there is STM32-RS needs to do a PAC release. The Github version has it, but I can't release this crate with a git dependency. 

I recommend posting [an issue here](https://github.com/stm32-rs/stm32-rs/issues), requesting a release for H5 support.

astapleton, 2023-07-22:
Ah, thanks for the quick response. I created an issue: https://github.com/stm32-rs/stm32-rs/issues/862

astapleton, 2023-07-22:
> Hi. It's untested, with a mix of peripheral support. So, you will almost surely find bugs that may or may not be easy to patch.
> 
> What it requires: Testing for a given use case, then squashing bugs and adding periph support as required.

Is there a test suite that you recommend putting it through once the stm32h5 PAC release is ready? I'm pretty new to rust, this repo, this processor, but would be interested in helping to fix some bugs if I can.

David-OConnor, 2023-07-22:
Build what you want to, and if something isn't working, we'll deal with it.

closed 2024-02-22 by David-OConnor