David-OConnor/stm32-hal #95: [STM32F401] USART6, required traits not implemented

stijn577, 2024-01-22:
### STM32F401 USART6, required traits not implemented 
I was trying to migrate from STM32f4xx-hal to this HAL, but when I tried to use USART6, it complained about traits not being implemented, see image.

[image]

```rs
let pp = ctx.device; // unwrapping the peripherals like this because of using rtic
let usart = Usart::new(pp.USART6, 115_200, UsartConfig::default(), &CLOCK);
```

I had a look in the source files of the repo at util.rs where this stuff is defined and indeed, USART6 is not defined for the STM32F401 board im using, I suspect this is a bug. Just posting it here to confirm, USART1 works fine (since the traits are implemented there, so that checks out).

David-OConnor, 2024-01-23:
Try now, using the Github dependency.

stijn577, 2024-01-23:
It seems its having an error, in the examples/syntax_overview/cargo.toml from the git dependency, the name contains spaces, which isnt allowed. So it might not actually changing to the git dependency. I haven't worked with git dependencies yet so idk if I'm doing something wrong. Also I'm in the middle of my exams right now so my response times might be a little slow, my apologies for that.

in my cargo.toml I have it set up like this:
```toml
[dependencies.stm32-hal2]
git = "https://github.com/David-OConnor/stm32-hal.git"
branch = "main"
features = ["f401", "f4rt"]
```
[image]

David-OConnor, 2024-02-22:
Example fixed.

closed 2024-02-22 by David-OConnor