David-OConnor/stm32-hal #96: Timer Interrupt Reson/Flag

revilo196, 2024-01-31:
Hi,

If having multiple Interrupts active on a Timer, like Update and multiple Channels, is there a way to check which interrupt(s) are active once inside the Interrupt function. 

I want to trigger different actions on different Channels of my timer.  

Is there anything i've overlooked? I can't find any function to get this information.

David-OConnor, 2024-01-31:
Try `my_timer.regs.sr().read().bits()`, and see which flags are set. Or `my_timer.regs.sr().read().uif.bit_is_set()` etc on a specific flag.

Note: Most periphs have a `read_status()` method to simplify this, but it's missing on Timer; adding for future version.

David-OConnor, 2024-02-15:
Closing due to inactivity, and likely solution.

closed 2024-02-15 by David-OConnor