David-OConnor/stm32-hal #99: [STM32G431KBT] Can't flash quickstart

evilGitHub, 2024-02-22:
I am trying to get the empty "stm32-hal-quickstart1" running on a Nucleo 32 G431 board. But the flashing process does not succeed. Is this something I can get help here or is this an issue for probe-rs?

- I have a STM32G431 Nucleo32 Board. (Chip is G431KBT)
- I updated the firmware of the onboard STLINK-V3E to the newest version.
- I installed the newest driver for the STLINK.
- If i run the command "probe-rs list" i get the following restult: `The following debug probes were found:
[0]: STLink V3 (VID: 0483, PID: 374e, Serial: 003C00173137510639383538, StLink)`
- I configured the project according to the readme.
- If i run the command "cargo run --release" the exaple is compiled, it tries to flash but i get the following error: `Error: The flashing procedure failed for 'target\thumbv7em-none-eabihf\release\project_name'.`

Any advice is appreciated :)

Full output is:
Compiling defmt-parser v0.3.4
   Compiling stm32-hal2 v1.8.1
   Compiling project_name v0.1.0 (C:\_Alles\_Git\cryptems\tobi\rustM4f_test\stm32-hal-quickstart1)
warning: unused import: `defmt::println`
  --> src\main.rs:13:5
   |
13 | use defmt::println;
   |     ^^^^^^^^^^^^^^
   |
   = note: `#[warn(unused_imports)]` on by default

warning: unused imports: `low_power`, `pac`
  --> src\main.rs:19:17
   |
19 | use hal::{self, low_power, pac};
   |                 ^^^^^^^^^  ^^^

warning: unused imports: `PinMode`, `Pin`, `Port`, `self`
 --> src\init.rs:5:11
  |
5 |     dma::{self, Dma},
  |           ^^^^
6 |     gpio::{Pin, PinMode, Port},
  |            ^^^  ^^^^^^^  ^^^^

warning: unused imports: `DmaInput`, `DmaInterrupt`, `OutputSpeed`, `PinMode`, `Pin`, `Port`, `self`
 --> src\setup.rs:4:11
  |
4 |     dma::{self, DmaInput, DmaInterrupt},
  |           ^^^^  ^^^^^^^^  ^^^^^^^^^^^^
5 |     gpio::{OutputSpeed, Pin, PinMode, Port},
  |            ^^^^^^^^^^^  ^^^  ^^^^^^^  ^^^^

warning: unused variable: `cp`
  --> src\init.rs:15:9
   |
15 |     let cp = cortex_m::Peripherals::take().unwrap();
   |         ^^ help: if this is intentional, prefix it with an underscore: `_cp`
   |
   = note: `#[warn(unused_variables)]` on by default

warning: struct `SystemStatus` is never constructed
 --> src\system_status.rs:1:12
  |
1 | pub struct SystemStatus {}
  |            ^^^^^^^^^^^^
  |
  = note: `#[warn(dead_code)]` on by default

warning: `project_name` (bin "project_name") generated 6 warnings (run `cargo fix --bin "project_name"` to apply 5 suggestions)
    Finished release [optimized + debuginfo] target(s) in 24.86s
     Running `probe-rs run --chip STM32G431KBT target\thumbv7em-none-eabihf\release\project_name`
      Erasing ✔ [00:00:02] [-----------------------------------------------------------------------------------------------------------------------------------]      0 B/10.00 KiB @ 4.97 KiB/s (eta 0s )
  Programming ✔ [00:00:02] [------------------------------------------------------------------------------------------------------------------------------------]      0 B/     0 B @      0 B/s (eta 0s )Error: The flashing procedure failed for 'target\thumbv7em-none-eabihf\release\project_name'.

Caused by:
    0: The initialization of the flash algorithm failed.
    1: Something during the interaction with the core went wrong
    2: A timeout occurred.
error: process didn't exit successfully: `probe-rs run --chip STM32G431KBT target\thumbv7em-none-eabihf\release\project_name` (exit code: 1)

David-OConnor, 2024-02-22:
Can you please confirm the following:

Your `.cargo/config.toml` looks like this:
```toml
[target.'cfg(all(target_arch = "arm", target_os = "none"))']
runner = "probe-rs run --chip STM32G431CBUx" # to list chips, run `probe-run --list-chips.`
rustflags = [
  "-C", "linker=flip-link",
  "-C", "link-arg=-Tlink.x",
  "-C", "link-arg=-Tdefmt.x",
  "-C", "link-arg=--nmagic",
]

[build]
target = "thumbv7em-none-eabihf"
```

memory.x like this:
```c
MEMORY
{
  /* 128K of flash; last 2k is a page for config. We limit it here since we're getting
   close to the limit, and want to catch spilling into it at link time. */
  FLASH : ORIGIN = 0x08000000, LENGTH = 126K
  RAM : ORIGIN = 0x20000000, LENGTH = 32K
}
```

Cargo.TOML dependencies section looks like this or similar:
```toml
[dependencies]
defmt = "^0.3.5"
defmt-rtt = "^0.4.0"
panic-probe = { version = "^0.3.0", features = ["print-defmt"] }

cortex-m = {version = "^0.7.7", features = ["critical-section-single-core"]}
cortex-m-rt = "0.7.3"
critical-section = "^1.1.2"

hal = { package = "stm32-hal2", version = "^1.8.1", features = ["g431", "g4rt", "usb", "can_fd_g"]}
```

evilGitHub, 2024-02-23:
@David-OConnor Thank you for your answer. But that did not fix it. I tried with your settings but I think they are not correct for me as I have a G431KBTx and not a G431CBUx. So I tried yours and I tried this (both with same result):

.cargo/config.toml :

```TOML
[target.'cfg(all(target_arch = "arm", target_os = "none"))']
# Change this runner as required for your MCU.
runner = "probe-rs run --chip STM32G431KBTx --probe 0483:374e --speed 1000" # to list chips, run `probe-rs chip list.`

rustflags = [
  "-C", "linker=flip-link",
  "-C", "link-arg=-Tlink.x",
  "-C", "link-arg=-Tdefmt.x",
  # This is needed if your flash or ram addresses are not aligned to 0x10000 in memory.x
  # See https://github.com/rust-embedded/cortex-m-quickstart/pull/95
  "-C", "link-arg=--nmagic",
]

[build]
# Change this target as required for your MCU.
target = "thumbv7em-none-eabihf" # Cortex-M4F and Cortex-M7F (eg F, L4, G4, H7)
# target = "thumbv6m-none-eabi"    # Cortex-M0 and Cortex-M0+ (eg G0)
# target = "thumbv8m.main-none-eabihf" # Cortex-M33F and Cortex-M35F (eg L5, U5, H5)

[alias]
rb = "run --bin"
rrb = "run --release --bin"
rr = "run --release"
br = "build --release"
```

memory.x :

```c
MEMORY
{
  FLASH : ORIGIN = 0x08000000, LENGTH = 128K
  RAM : ORIGIN = 0x20000000, LENGTH = 22K
}
```

Cargo.TOML :

```TOML
[dependencies]
defmt = "0.3.5"
defmt-rtt = "0.4.0"
panic-probe = { version = "0.3.0", features = ["print-defmt"] }

cortex-m = {version = "^0.7.7", features = ["critical-section-single-core"]}
cortex-m-rt = "0.7.3"
critical-section = "^1.1.2"

#rtic = { version = "^2.0.1", features = ["thumbv7-backend"] }

# Change this import as required for your MCU.
hal = { package = "stm32-hal2", version = "^1.8.1", features = ["g431", "g4rt"]}
```

Also I tired without the --probe and without the --speed option. I did confirm that I can flash this board with OpenOCD with a c program that is then executed. So board is not broken. 

Additionally I tried this command:
`probe-rs erase --chip STM32G431KBTx --probe 0483:374e --speed 1000`
which resulted in the following output: 
`Error: The execution of 'init' failed with code 134217728. This might indicate a problem with the flash algorithm.`

David-OConnor, 2024-02-23:
Weird! No idea. Of note, I use the G431 (CBU) with ST-LinkV3 in lots of projects. If it's failing to flash, it likely has something to do with one of the configuration sections I posted.

Something else that *may* be related. Can you confirm you have these lines in `main.rs`?

```rust
use defmt_rtt as _;
use panic_probe as _;

#[defmt::panic_handler]
fn panic() -> ! {
    cortex_m::asm::udf()
}

```

evilGitHub, 2024-02-23:
I used the empty example from here. I can confirm I have the lines you asked for. Also tried removing the watchdog setup line. Made no difference. 

I also tried a second Nucleo Board with the same chip and had no luck either. At last I tried a  Nucleo L432KC Board (different chip) and run the "probe-rs erase" command that worked. So it is either related to the Chips or the boards. I will post a issue at probe-rs repository and report back here if I get a solution from there.

Thank you for your time :)

David-OConnor, 2024-02-23:
LMK what the problem ends up as!

evilGitHub, 2024-02-23:
https://github.com/probe-rs/probe-rs/issues/2222

evilGitHub, 2024-02-24:
This was caused a known bug in probe-rs which is already fixed in master branch but not released yet. Compiling and installing probe-rs from head of master solved this. Now this is working fine :) thanks again for your assistance!

closed 2024-02-24 by evilGitHub