David-OConnor/stm32-hal #100: [STM32H7A3] Can't compile

cracksalad, 2024-04-09:
I am new to Rust for STM32, so please excuse any stupid mistakes.

When I cloned the quickstart and configured it, the following errors show up during compilation:

```
   Compiling stm32-hal2 v1.8.2
error[E0433]: failed to resolve: could not find `ADC3_COMMON` in `pac`
    --> ~\.cargo\registry\src\index.crates.io-6f17d22bba15001f\stm32-hal2-1.8.2\src\adc.rs:1236:20
     |
1236 |         hal!(ADC3, ADC3_COMMON, adc3, 3);
     |                    ^^^^^^^^^^^
     |                    |
     |                    could not find `ADC3_COMMON` in `pac`
     |                    help: a struct with a similar name exists: `ADC12_COMMON`

error[E0412]: cannot find type `ADC3` in module `pac`
    --> ~\.cargo\registry\src\index.crates.io-6f17d22bba15001f\stm32-hal2-1.8.2\src\adc.rs:1236:14
     |
1236 |         hal!(ADC3, ADC3_COMMON, adc3, 3);
     |              ^^^^ help: a struct with a similar name exists: `ADC1`
     |
    ::: ~\.cargo\registry\src\index.crates.io-6f17d22bba15001f\stm32h7-0.15.1\src\stm32h7b3\mod.rs:684:1
     |
684  | pub struct ADC1 {
     | --------------- similarly named struct `ADC1` defined here

error[E0412]: cannot find type `SAI3` in module `pac`
    --> ~\.cargo\registry\src\index.crates.io-6f17d22bba15001f\stm32-hal2-1.8.2\src\util.rs:593:25
     |
593  | impl RccPeriph for pac::SAI3 {
     |                         ^^^^ help: a struct with a similar name exists: `SAI1`
     |
    ::: ~\.cargo\registry\src\index.crates.io-6f17d22bba15001f\stm32h7-0.15.1\src\stm32h7b3\mod.rs:2896:1
     |
2896 | pub struct SAI1 {
     | --------------- similarly named struct `SAI1` defined here

error[E0412]: cannot find type `SAI4` in module `pac`
    --> ~\.cargo\registry\src\index.crates.io-6f17d22bba15001f\stm32-hal2-1.8.2\src\util.rs:620:25
     |
620  | impl RccPeriph for pac::SAI4 {
     |                         ^^^^ help: a struct with a similar name exists: `SAI1`
     |
    ::: ~\.cargo\registry\src\index.crates.io-6f17d22bba15001f\stm32h7-0.15.1\src\stm32h7b3\mod.rs:2896:1
     |
2896 | pub struct SAI1 {
     | --------------- similarly named struct `SAI1` defined here
```

Refering to the [device coverage of stm32-rs](https://stm32-rs.github.io/stm32-rs/STM32H7B3x.html) there is no `ADC3`, `SAI3` and `SAI4`. 

My quickstart config looks like this:

- Cargo.toml: `hal = { package = "stm32-hal2", version = "^1.6.4", features = ["h7b3", "h7rt"]}`
- .cargo/config.toml: `runner = "probe-rs run --chip git STM32H7A3ZITxQ"` and `target = "thumbv7em-none-eabihf"`
- memory.x: `FLASH : ORIGIN = 0x08000000, LENGTH = 1953K` and `RAM : ORIGIN = 0x20000000, LENGTH = 1367K` (really not sure if this is correct, but should not be the problem here)

There seems to be a mismatch between the target chip and the code somehow. Any hints?

David-OConnor, 2024-04-09:
Yep, that's exactly what it is. The examples are general, while this lib supports many STM32 variants. So, you'll have to modify the peripherals used as required based on which you have, and which are wired up.

cracksalad, 2024-04-10:
@David-OConnor thanks for confirming! Do you have any documentation about how to configure this?

cracksalad, 2024-07-04:
@David-OConnor is it as easy as adding a `not(feature = "h7b3")` to those snippets:

https://github.com/David-OConnor/stm32-hal/blob/b9a742157c50d3fe6e22ec7e000c722d223154ce/src/adc.rs#L1231-L1238

https://github.com/David-OConnor/stm32-hal/blob/b9a742157c50d3fe6e22ec7e000c722d223154ce/src/util.rs#L592-L593

https://github.com/David-OConnor/stm32-hal/blob/b9a742157c50d3fe6e22ec7e000c722d223154ce/src/util.rs#L619-L620

David-OConnor, 2024-07-04:
Nice!