David-OConnor/stm32-hal #103: calc_speeds() method in your clock_cfg.rs example ?

ffred, 2024-06-15:
Hello,
first test with Rust embedded. just tried to keep the minimum from your clock_cfg.rs example, with the default clk settings, but 
I got this error :  _no method named `calc_speeds` found for struct `Clocks` in the current scope_

did I miss something in my config or what ?
(I created my project from your quickstart one...)
thanks

```
#![no_main]
#![no_std]

use cortex_m_rt::entry;
use hal::{
    clocks::Clocks,
    low_power, pac,
};

#[entry]
fn main() -> ! {
    // Set up CPU peripherals
    let mut cp = cortex_m::Peripherals::take().unwrap();
    // Set up microcontroller peripherals
    let mut dp = pac::Peripherals::take().unwrap();

    // Set up a default setting. 
    let mut clock_cfg = Clocks::default();

    // Configure clock registers.
    clock_cfg.setup().unwrap();

    // Show speeds.
    defmt::println!("Speeds: {:?}", clock_cfg.calc_speeds());

    loop {
        low_power::sleep_now();
    }
}

#[defmt::panic_handler]
fn panic() -> ! {
    cortex_m::asm::udf()
}
```

David-OConnor, 2024-06-15:
That example is out of date; the method no longer exists.

ffred, 2024-06-15:
Ok, thanks..

closed 2024-06-15 by ffred