David-OConnor/stm32-hal #116: Quickstart fails to compile for STM32L552ZE

Ictogan1, 2024-12-18:
Wanted to build the quickstart repo for the STM32L552ZE today, failed with the following errors.

```
error[E0599]: no method named `write` found for struct `BitWriterRaw` in the current scope
   --> /home/markus/.cargo/registry/src/index.crates.io-6f17d22bba15001f/stm32-hal2-1.8.4/src/i2c.rs:460:24
    |
460 |             w.rd_wrn().write();
    |                        ^^^^^ method not found in `BitWriterRaw<'_, u32, CR2_SPEC, bool, BitM, 10>`

error[E0599]: no method named `read` found for struct `BitWriterRaw` in the current scope
   --> /home/markus/.cargo/registry/src/index.crates.io-6f17d22bba15001f/stm32-hal2-1.8.4/src/i2c.rs:492:24
    |
492 |             w.rd_wrn().read();
    |                        ^^^^ method not found in `BitWriterRaw<'_, u32, CR2_SPEC, bool, BitM, 10>`

error[E0599]: no method named `is_busy` found for struct `BitReaderRaw` in the current scope
   --> /home/markus/.cargo/registry/src/index.crates.io-6f17d22bba15001f/stm32-hal2-1.8.4/src/i2c.rs:531:43
    |
531 |         while self.regs.isr.read().busy().is_busy() {}
    |                                           ^^^^^^^ method not found in `BitReaderRaw<bool>`

For more information about this error, try `rustc --explain E0599`.
error: could not compile `stm32-hal2` (lib) due to 3 previous errors
```

Changes from baseline repo I've made:
```
+++ b/.cargo/config.toml
@@ -1,6 +1,6 @@
 [target.'cfg(all(target_arch = "arm", target_os = "none"))']
 # Change this runner as required for your MCU.
-runner = "probe-rs run --chip STM32H723ZGTx" # to list chips, run `probe-rs chip list.`
+runner = "probe-rs run --chip STM32L552ZE" # to list chips, run `probe-rs chip list.`
 
 rustflags = [
   "-C", "linker=flip-link",
@@ -13,9 +13,9 @@ rustflags = [
 
 [build]
 # Change this target as required for your MCU.
-target = "thumbv7em-none-eabihf" # Cortex-M4F and Cortex-M7F (eg F, L4, G4, H7)
+#target = "thumbv7em-none-eabihf" # Cortex-M4F and Cortex-M7F (eg F, L4, G4, H7)
 # target = "thumbv6m-none-eabi"    # Cortex-M0 and Cortex-M0+ (eg G0)
-# target = "thumbv8m.main-none-eabihf" # Cortex-M33F and Cortex-M35F (eg L5, U5, H5)
+target = "thumbv8m.main-none-eabihf" # Cortex-M33F and Cortex-M35F (eg L5, U5, H5)
 
 [alias]
 rb = "run --bin"
diff --git a/Cargo.toml b/Cargo.toml
index 77b982e..0c31562 100644
--- a/Cargo.toml
+++ b/Cargo.toml
@@ -16,7 +16,7 @@ critical-section = "^1.1.2"
 #rtic = { version = "^2.0.1", features = ["thumbv7-backend"] }
 
 # Change this import as required for your MCU.
-hal = { package = "stm32-hal2", version = "^1.8.4", features = ["l4x3", "l4rt"]}
+hal = { package = "stm32-hal2", version = "^1.8.4", features = ["l5rt", "l552"]}
 
 # cargo build/run
 [profile.dev]
diff --git a/memory.x b/memory.x
index 5f1a99b..6c1b4de 100644
--- a/memory.x
+++ b/memory.x
@@ -3,6 +3,6 @@
 MEMORY
 {
   /* NOTE 1 K = 1 KiBi = 1024 bytes */
-  FLASH : ORIGIN = 0x08000000, LENGTH = 256K
-  RAM : ORIGIN = 0x20000000, LENGTH = 64K
+  FLASH : ORIGIN = 0x08000000, LENGTH = 512K
+  RAM : ORIGIN = 0x20000000, LENGTH = 256K
 }
```

David-OConnor, 2024-12-18:
This is a bug in the latest i2c version affecting all L5 variants. Will attempt to fix when I find time.

David-OConnor, 2024-12-19:
Try now