David-OConnor/stm32-hal #119: I2C address is zero for write and write_read on WB55

alexheslop, 2025-01-05:
Hi,

Thanks for all the hard work on this HAL, it's been excellent using it so far. I've run into an issue trying to use the I2C on a WB55 Nucleo Board, which I wondered whether you may be able to assist with. 

I am seeing an address of 0x00 being no matter what the value I put into the i2c.write or i2c.write_read functions.

Here is my code:

`#![no_std]
#![no_main] // Don't use the Rust standard library, don't use the Rust entry point

use cortex_m::delay;
use hal::{
    clocks::Clocks,
    gpio::{Pin, PinMode, Port},
    i2c::{I2c, I2cConfig},
    pac::{self, interrupt, CorePeripherals},
};

use panic_halt as _;
use rtt_target::{rprintln, rtt_init_print};

#[entry]
fn main() -> ! {
    rtt_init_print!();

    let cp: CorePeripherals = cortex_m::Peripherals::take().unwrap();
    let dp: pac::Peripherals = pac::Peripherals::take().unwrap();

    let clock_cfg: Clocks = Clocks::default();
    clock_cfg.setup().unwrap();

    let mut delay = delay::Delay::new(cp.SYST, clock_cfg.systick());

    let mut scl = Pin::new(Port::B, 8, PinMode::Alt(4));
    scl.output_type(hal::gpio::OutputType::OpenDrain);
    scl.output_speed(hal::gpio::OutputSpeed::VeryHigh);
    scl.pull(hal::gpio::Pull::Floating);

    let mut sda = Pin::new(Port::B, 9, PinMode::Alt(4));
    sda.output_type(hal::gpio::OutputType::OpenDrain);
    sda.output_speed(hal::gpio::OutputSpeed::VeryHigh);
    sda.pull(hal::gpio::Pull::Floating);

    let i2c_config = I2cConfig {
        speed: hal::i2c::I2cSpeed::Fast400K,
        mode: hal::i2c::I2cMode::Master,
        address_bits: hal::i2c::AddressBits::B7,
        nostretch: false,
        smbus: false,
        noise_filter: hal::i2c::NoiseFilter::Analog,
    };

    let mut i2c: I2c<pac::I2C1> = I2c::new(dp.I2C1, i2c_config, &clock_cfg);
    delay.delay_ms(500);

    loop {
        if i2c.write(0x19, &[0x0F]).is_ok() {
            rprintln!("Write to address: 0x19");
        } else {
            rprintln!("Write failed");
        }

        delay.delay_ms(500);
    }
}
`

and a screenshot from a Picoscope with the serial decoding:

[image]

It's my first time using this HAL so it's certainly feasible that it's a user error! Thanks in advance.

David-OConnor, 2025-01-14:
Great info; I appreciate it! I'll take a look, and QC. I do have a WB dev board lying around.

David-OConnor, 2025-05-21:
Could you please try with the latest version, and see if it's fixed? I'm reasonably sure it is; we had a I2C rewrite that broke it; reverted now.

David-OConnor, 2025-05-25:
Closing as likely fixed.