David-OConnor/stm32-hal #123: STM32F4 USART Support

lindblandro, 2025-02-01:
I noticed that the USART support for stm32f4 is somewhat limited as it doesn't cover interrupts or dma transfers. The information is buried at the very end of the repository readme, in the errata section. I think interrupts / dma are essential for any real project and might be a deal breaker for stm32-hal (if one has to work with stm32f4).

lindblandro, 2025-02-01:
I can probably take a swing at adding interrupt / dma support, but I only have a nucleo-stm32f429 for verification. I also took a preliminary look at the code and there is a lot of conditional compilation going on. There is some effort done in generalizing access to the interrupt status register (inconveniently named "states register" on the f4), but all in all the f4 behaves a little differently, starting from the available interrupts, and there's probably going to be a lot more `#cfg` blocks. Any ideas on how to manage those?

David-OConnor, 2025-02-04:
I'm with you on interrupts and DMA being essential. Bottom line: I consider F4 to be a legacy MCU I only added because it was popular in the rust community, and hit an obstacle etc. I'll merge any PRs that address this.

lindblandro, 2025-02-05:
Yeah it's probably not very common nowadays. I have a couple of existing designs that have an F4 and nucleo-f429 has been my go to devkit for some time now. For newer stuff the U5 (for security) and / or one of the chips you mentioned (for basic stuff) are probably the way to go in the future, with CRA coming into effect and all. 

I can understand the obstacles and not giving F4 much thought. After comparing the F4 and even F3 I noticed that the F4 is just weird, the flash sectors for example are all different size for some reason, and the U(S)ART peripheral clearly is older than in the other STMs and requires special handling. 

I'll try to find the time to properly test my PR in the coming days.