David-OConnor/stm32-hal #127: STM32G031 PWM output doesn't set output value (using set_duty)

araneusdiadema, 2025-03-10:
When trying to set output PWM value on STM32G031 using "set_duty" function I consistently got no output and when reading the value using "get_duty" I always get 0. After searching through the "timers.rs" file I found that section in "set_duty" corresponding to the "g0" is commented out. Not sure if it's a typo or it's there on purpose and the issue is in something else."

David-OConnor, 2025-06-08:
Hey! I don't remember exactly, but I likely hit some sort of PAC-related compile error on this. Open to PRs if you think you can get it working! (e.g. uncomment it, compile, and see what the errors are). Likely not a big blocker.

David-OConnor, 2025-12-06:
Note: I believe it currently panics. But still not implemented.