David-OConnor/stm32-hal #135: Support for the C0 series

FeldrinH, 2025-07-07:
Are there any plans to support the STM32C0xx series? I'm particularly interested in STM32C071 for its USB capabilities.

David-OConnor, 2025-07-07:
I would love C0 support, but don't have any hardware, and not sure what state the C0 PAC is in. WOuld accept a PR, even if limited functionality, e.g. only a few peripherals supported. Is it most similar to G0, you think?

FeldrinH, 2025-07-07:
I've heard it is similar to G0, but I have not compared the reference manuals myself.

FeldrinH, 2025-07-07:
As for the PAC I'm not really sure how to evaluate it's maturity, since it's autogenerated. It does exist, it supports STM32C011, STM32C031 and STM32C071 and has the USB registers for C071.

David-OConnor, 2025-07-07:
Obstacle: c0 was PAC was added as 0.16.0. (The latest as of a month ago). 

In order to conserve most of the code, we'd probably have to update the others as well. (They're on v15, the prev release, a few years old). This may be a big undertaking I hoped to avoid. In particular, there are a lot of cases where the code is the same, but they made fields private and exposed methods. So a lot of adding `()`... all over the code base.

FeldrinH, 2025-07-07:
> In order to conserve most of the code, we'd probably have to update the others as well. (They're on v15, the prev release, a few years old). This may be a big undertaking I hoped to avoid. In [...]

That is indeed quite annoying, but the update probably needs to be done eventually, so might be good to get it out of the way now.

I'll fiddle around a bit and see if I can automate the more rote parts of the migration.

FeldrinH, 2025-07-09:
By the way, do you have some kind of test suite or documented testing procedure for testing the HAL on new hardware?

David-OConnor, 2025-07-09:
What peripherals are on the C0 board you're using, and which variant is it? I will focus on those so you can test it.

FeldrinH, 2025-07-09:
The variant is STM32C071RB. Was planning to buy a NUCLEO-C071RB devboard or something equivalent. I don't know all the peripherals but the ones we will use are the USB (as device), SPI and GPIO.

David-OConnor, 2025-07-09:
Here is the rub:

Generally, things get in a good state for MCUs and peripherals that someone has used, and either cross-checked the manual with, or tried and found/fixed problems. Or, for peripherals that are the same as one that has been tested as such. So, G4 and H7 are very well supported because I've done a lot of work on them using most of their periphs. 

I think the best way to work this is to focus on getting the basics (RCC, Flash memory, and GPIO) working on C0, through some trial and error, then getting SPI and USB working. I think this will be easy since that's a narrow feature set.

FeldrinH, 2025-07-09:
> I don't know all the peripherals but the ones we will use are the USB (as device), SPI and GPIO.

Oh, and we'll be using PWM as well.

David-OConnor, 2025-07-09:
The latest push has those peripherals added for C0, but does not compile. In addition to getting it to compile, I think the clocks (clocks/baseline.rs) needs a RM review: It appears we're  missing most of the PLL-related functionality in the PAC, and I haven't looked into why.

FeldrinH, 2025-07-09:
> It appears we're missing most of the PLL-related functionality in the PAC, and I haven't looked into why.

What exactly do you mean? I checked the STM32C0 reference manual and I could not find any PLL related functionality there.

David-OConnor, 2025-07-09:
Exactly. Tha'ts a first for this lib. Maybe it's jsut a matter of feature-gating out all the PLL code, but I don't know.

FeldrinH, 2025-07-09:
Found this in some random STM presentation:

[image]

([source](https://www.st.com/content/ccc/resource/training/technical/product_training/group1/99/8d/2e/07/1c/1a/43/33/stm32c0-system-reset-and-clock-controller/files/stm32c0-system-reset-and-clock-controller.pdf/jcr:content/translations/en.stm32c0-system-reset-and-clock-controller.pdf))

David-OConnor, 2025-07-10:
Looks like it just runs off the HSI directly. Probably a matter of just feature-gating out the PLL lines.

FeldrinH, 2025-07-15:
RccPeriph is not implemented for SPI2 and I2C2 despite being available in the STM32C0 PAC. Is there a reason RccPeriph is not implemented for them?

David-OConnor, 2025-07-15:
No. You should assume that there is no good reason for any decisions I've made regarding C0; it's full of low-effort best-guesses!

FeldrinH, 2025-07-17:
> WRT this TODO note: https://github.com/David-OConnor/stm32-hal/blob/main/src/clocks/baseline.rs#L1138. I believe the PAC is fine, it's just that the fields are `hsiusb48on` and `hsiusb48rdy`

I patched this in my local code and now I am getting events, but Windows still detects the device as invalid. More specifically, the device gets suspended, then resumed and then gets some data. After this there is no more events and Windows complains that the device descriptor request failed.

David-OConnor, 2025-07-17:
Good find. I updated the repo with that fix, and enabled the `hsi48_on` setting in `Clocks`. Note that this defaults to False.

Beyond clocks, the main things to check are if the USB ram is configure correctly in the `usb` module. I did look it up and add, but didn't review that module thoroughly.

Would you mind posting your USB code? Something to keep in mind is that the PC will usually drop the connection if it doesn't respond quickly to queries. I'm guessing this isn't the problem, but worth checking. (If you have an interrupt that runs `poll`, or a poll loop while testing, that will resolve that problem)

In general, I'm also curious how other peripherals are working, e.g. GPIO, RTC, IDWG, or any busses that are testable with that dev board. But maybe one thing at a time... 

I'm also curious if the system clock speed is running at the expected speed.

FeldrinH, 2025-07-17:
> Good find. I updated the repo with that fix, and enabled the `hsi48_on` setting in `Clocks`.

What's up with the ` #[cfg(not(feature = "c0"))]` [here](https://github.com/David-OConnor/stm32-hal/blob/main/src/clocks/baseline.rs#L1142)? Also, shouldn't the while loop wait for `hsiusb48rdy` instead of `hsiusb48on`?

FeldrinH, 2025-07-17:
> Beyond clocks, the main things to check are if the USB ram is configure correctly in the `usb` module. I did look it up and add, but didn't review that module thoroughly.
>
> Would you mind posting [...]

Will get to these over the coming days.

David-OConnor, 2025-07-18:
Good find.

Given that this is probably an easy patch, I think this is the move. (I've done something similar with that lib before):

- Fork it and make the change
- Send in the PR
- Release the fork on crates.io, under a slightly different name.
- In this HAL, add the patched crate as an `optional` dependency, and have it be used in the case of C0, vice the upstream.

We do the last step, as the stm32-usbd import is required in firmware too, and we don't want to force non-C0 firmwares to use this temp lib. When the stm32-usbd maintainers merge and release, we point back to upstream.

If you do the usbd patch and release, I'll update this HAL IOC that, if you like this plan.

FeldrinH, 2025-07-18:
I'll make the PR, but maybe you could release it under your existing stm32-usbd2 crate. I would rather not pollute crates.io with more temporary crates if possible.

FeldrinH, 2025-07-19:
I have opened the PR: https://github.com/stm32-rs/stm32-usbd/issues/37. I tested it on my C071 and everything seemed to work.

Also, quick question about clock configuration: how do I set SYSDIV and HSIDIV?

David-OConnor, 2025-07-20:
[Published](https://crates.io/crates/stm32-usbd2)

Updated this lib to use if for C0

Added HsiDiv and Sysdiv (C071 only for the latter) to the clock config. Set them to default to hardware defaults of hsidiv = 4, sysdiv = 1. Thoughts on these defaults? Anything I missed?

FeldrinH, 2025-07-20:
> Added HsiDiv and Sysdiv (C071 only for the latter) to the clock config. Set them to default to hardware defaults of hsidiv = 4, sysdiv = 1. Thoughts on these defaults?

I think matching hardware defaults is sensible. Less suprises when working off of the reference manual.

FeldrinH, 2025-07-20:
> Updated this lib to use if for C0

The HAL is currently using the old version of stm32-usbd2 (0.7.0), so it's not using my patch.

David-OConnor, 2025-07-20:
Fixed. Is this right?

```rust
    #[cfg(any(feature = "c0", feature = "h5"))]
    const EP_MEMORY_ACCESS: MemoryAccess = MemoryAccess::Word32x1;
```

David-OConnor, 2025-07-20:
Done

David-OConnor, 2025-07-20:
> > Added HsiDiv and Sysdiv (C071 only for the latter) to the clock config. Set them to default to hardware defaults of hsidiv = 4, sysdiv = 1. Thoughts on these defaults?
> 
> I think matching hardware defaults is sensible. Less suprises when working off of the reference manual.

For some context on my Q, we don't use hardware defaults for most MCUs: The hardware usually defaults to 16Mhz HSI (on most variants). Our defaults enable the PLL, and configure to the max clock speed using it. C0 is different not having a PLL, and I haven't looked into if it inits to max clock speed (The HSIDIV we just added may be relevant here). So, a bit of new territory here.

FeldrinH, 2025-07-20:
I've just started testing SPI support. Quick question about SPI configuration: How should I determine the appropriate `fifo_reception_thresh` value? The reference manual says that "The RXFIFO threshold must be aligned to the read
access size for the SPIx_DR register.", but all the read access to SPIx_DR happens inside the HAL, so why doesn't the HAL set the threshold automatically?

FeldrinH, 2025-07-21:
Also, quick observations from testing some clocks:
* SYSTICK calculation is wrong for multiple families. This is kind of poorly documented, but AFAIK it should be HCLK / 8 if the clock source in the SYST_CSR register is set to external clock (the default) and HCLK otherwise (this is true for at least the C0, G0 and H7).
* SYSCLK calculation is wrong for C0. It should be divided by SYSDIV in all cases and additionally divided by HSIDIV if the source is HSI. Also the HSI base frequency on C0 is 48 MHz, not 16 MHz.

FeldrinH, 2025-07-21:
Also, I'm a bit confused by the existence of the `hsi48_on` flag. Other clock sources are automatically enabled when they are used as an input for something. Why is does HSI48 need an extra flag to be enabled? Couldn't it just be enabled automatically when used as a source for `clk48_src` and whatever else HSI48 can supply.

(Getting rid of `hsi48_on` would also resolve another annoyance: Currently the naming of the `hsi48_on` flag is quite confusing on C071, because C0 has a HSI48 clock, but `hsi48_on` is actually used to enable the HSI48USB clock, not the HSI48 clock.)

David-OConnor, 2025-07-21:
Yea; you're right on the systick() calculation method being incorrect in some cases. (Anywhere where it's not equal to hclk) I think it's correct for the most common cases, but not all. Would merge a PR to fix.

On variants other than C0, the hsi48_on flag is needed, at least in the current setup. I think what you're proposing is have it so clk48_src = Hsi48 automatically turns it on, but I think this doesn't make sense for the case of (and this is the default case) of not using USB/HSI-48. E.g., to have it off without the hsi48_on flag, you would need to set Clk48Src to a bogus source like Sai.

I also noticed the confusing names on C0 with Hsi48; not sure the best way to handle that for clarity. Probably stick with RM naming conventions?

Regarding pclk vs apb: You're correct that it's inconsistent. Overall, it's inconsistent throughout the reference material.

FeldrinH, 2025-07-21:
Also, `lpuart_src` and `sai1_src` fields are enabled on C0, but don't seem to do anything.

FeldrinH, 2025-07-21:
> Yea; you're right on the systick() calculation method being incorrect in some cases. (Anywhere where it's not equal to hclk) I think it's correct for the most common cases, but not all. Would merge a PR to fix.

Can you name a chip where the current systick() is definitely correct? I looked through quite a few reference manuals and they all seem to use HCLK / 8 for SYTICK by default.

David-OConnor, 2025-07-21:
lupart_src and sai1 removed from c0. good call.

Regarding Systick: I'm not sure, but suspect you're right. I think it can be either 1 or 8. (And is maybe configured in a Cortex-M reg, not an STM32 one??) I will provide this selection of info:

This helper fn in this HAL:
```rust
/// A blocking delay, for a specified time in ms.
pub fn delay_ms(num_ms: u32, ahb_freq: u32) {
    let cp = unsafe { cortex_m::Peripherals::steal() };
    let mut delay = Delay::new(cp.SYST, ahb_freq);
    delay.delay_ms(num_ms);
}
```
Works correctly when I provide the AHB frequency directly, e.g. set it to 170_000_000 on G4 at full speed.

The G4 RM is sparse on this, with this as the main info implying 0 or 8 divisor. Not sure which is default:

[image]

G4 clock tree. Note that the divider to Sys Timer can be either 1 or 8; I believe it defaults to 1, but don't have the source on that:

[image]

The H7 clock tree is the same, with a 1 or 8x divider.

Gut says it can be 1 or 8, probably on any stm32, and defaults to 1, but I'm not sure on that default.

FeldrinH, 2025-07-21:
It definitely defaults to 8 on C0. Note that `Delay::new` internally configures SysTick to use the core clock, which means a divider of 1 regardless of the default.

FeldrinH, 2025-07-21:
I believe it is controlled by the `CLKSOURCE` field on the [`SYST_CSR` register](https://developer.arm.com/documentation/dui0552/a/cortex-m3-peripherals/system-timer--systick/systick-control-and-status-register). The default for `CLKSOURCE` is 0 (external clock), if an external clock exists and 1 (core clock) otherwise. I believe all or at least the vast majority of STM32 devices have the external clock wired as HCLK / 8.

FeldrinH, 2025-07-21:
Unfortunately, reading the clock source can have an unwanted side effect (see the note here https://docs.rs/cortex-m/latest/cortex_m/peripheral/struct.SYST.html#method.get_clock_source).

David-OConnor, 2025-07-21:
Given my note on this not being used internally, how about we change it to hclk/8, and note our conclusions in its doc comment?

FeldrinH, 2025-07-21:
Honestly I don't know. It's really unfortunate that the hardware default and the setting used by cortex-m delay don't match. I looked at the other HAL's for insipiration and they don't seem to have a function to calculate systick frequency. Probably because of the same messiness that we have run into.

FeldrinH, 2025-07-21:
Or just recommend people to use `systick() / 8` if they are using systick with the hardware default. Most libraries that use systick seem to configure the clock source to core, so actually using the hardware default with its division by 8 seems to be the exception.

FeldrinH, 2025-07-22:
If I am reading the reference manual right it says that you should set FRXTH to 16-bit if you are doing 16-bit reads on RXFIFO and 8-bit if you are doing 8-bit reads on RXFIFO. The HAL seems to always do 8-bit reads, so FRXTH should probably be fixed to 8-bit and fifo_reception_thresh removed from the SPI configuration.

David-OConnor, 2025-07-23:
Sounds right. I think my original intent was to support both, but never got around to 16-bit support.

Unrelated - have you been able to confirm which periphs on C0 work and don't work? So far I'm tracking:

- USB works.

I suspect that GPIO reads and writes, RTC, and timers would be the most straightforward to test. Then any busses that connect to something on that dev board. I think we're through the worst of it with the clocks.

FeldrinH, 2025-07-23:
* The clocks seem to work, at least for the combinations I tried the actual behavior and my understanding matched. I can get 48 MHz if I set all dividers to 1. HSI and HSE do both work as inputs.
* GPIO input and output works (at least in simple cases, I tested with devboard LEDs and buttons).

The rest I haven't tested yet. I have code to test SPI and PWM, but lack the equipment to verify they are actually working (I am currently away from my home city, so I don't have a decent oscilloscope or logic analyzer).

FeldrinH, 2025-07-23:
PS: I don't know if this is intentional, but with https://github.com/David-OConnor/stm32-hal/commit/4b140b1aa150795585ee6fd1d70cdd622f47ea80, C071 now needs both `usb` and `usb_c0` features to enable USB support.

David-OConnor, 2025-07-23:
Thx. Not intentional/fixed I think on USB.

David-OConnor, 2025-07-26:
I can confirm this on G4 as well. It may not be what you're experiencing, but it's likely this is the case. I hadn't mentioned it because we have been chatting about C0, but this is currently my #1 priority for this library.

What's weird is I did a complete register dump and diff between 1.95 (Working SPI), and 2.0, just prior to the SPI failure to read, and didn't see any relevant differences! Which is so weird.

I suspect you've solved it, because that would explain the faulty behavior, and me being unable to reproduce it by comparing registers. I will try your fix. If it works, I will immediately release a patch version.

David-OConnor, 2025-07-26:
Verified, and released as 2.1.0. Great stuff; saved me lots of hassle.

FeldrinH, 2025-07-26:
Can confirm that SPI is working on C0

FeldrinH, 2025-07-27:
stm32-usbd released a new version

David-OConnor, 2025-07-27:
Updated here.

FeldrinH, 2025-07-27:
Should probably bump stm32-usbd version in dependencies. Also, C0 USB support is still gated by `usb_c0` feature which no longer exists.

FeldrinH, 2025-07-27:
Latest version doesn't compile on C071. Looking at the CI it doesn't seem to compile on most devices.

David-OConnor, 2025-07-27:
Fixed; looks like a change to the API of usbd

David-OConnor, 2025-08-06:
Yea tracking. I ave this in the Eratta. Are you saying this is happening on a non-Advanced timer on C0?

```
- If using PWM (or output compare in general) on an Advanced control timer (eg TIM1 or 8),
you must manually set the `TIMx_BDTR` register, `MOE` bit.
```

I don't remember why I put this in instead of just making the code work. My guess is that the Timer module doesn't distinguish between Advanced control and regular. (Although it does distinguish Basic, and doesn't support low power timers)

David-OConnor, 2025-08-07:
Yea, I think that would work.

FeldrinH, 2025-08-07:
> You're welcome at trying a PR; if it compiles, it's probably correct.

Sure, I'll take a crack at it. How much do you care about the minimum supported Rust version for this crate? In other words, can I use recently stabilized features?

David-OConnor, 2025-08-07:
Cleared hot to use recent features. I just updated another lib I made today to require a version that just came out, for context...