David-OConnor/stm32-hal #143: Why does set_input_capture not handle TISEL?

FeldrinH, 2025-07-16:
The documentation for Timer::set_input_capture says that the function does not handle setting TISEL registers. Why is that? Is configuring those registers very complex? Would be much nicer if I didn't have to reach into PAC to set up input capture.

David-OConnor, 2025-07-16:
I don't remember, but from a RM review, I suspect that it had to do with lots of PAC variation between variants, and across the different channels. It's worth taking another look at, especially with the updated PAC. (Which, for example, dramatically simplified channel-based code for DMA and DFSDM)

FeldrinH, 2025-07-29:
I'm interested in part of this functionality for my project. Specifically, I need to read a quadrature encoder.

Would you be open to a patch implementing encoder mode support and if so, what should the API look like? A particualr issue that I am thinking about is that when using encoder interface mode the usual frequency calculations don't make sense and can break things. This is troublesome because currently there is no way to construct a `Timer` without setting the frequency.

David-OConnor, 2025-07-29:
Would merge a PR. Don't know how this works myself.

To override the timer frequency, set ARR and PSC manually after construction.

FeldrinH, 2025-07-29:
Other HALs seem to have a separate Qei peripheral for encoder interface mode (e.g. [embassy_stm32::timer::qei::Qei](https://docs.embassy.dev/embassy-stm32/git/stm32f405vg/timer/qei/struct.Qei.html), [stm32f4xx_hal::qei::Qei](https://docs.rs/stm32f4xx-hal/latest/stm32f4xx_hal/qei/struct.Qei.html)). Maybe this HAL should adopt a similar API?

FeldrinH, 2025-08-02:
@David-OConnor What do you think?

David-OConnor, 2025-08-02:
I think it's a great idea, and would merge a PR. I'm not in a position to do this myself, as I have never used quadrature, don't know what it is, and have no use case for it to test with.

FeldrinH, 2025-08-02:
I created a PR (https://github.com/David-OConnor/stm32-hal/pull/145). @David-OConnor, I would like your input on the open questions, because they are API design decisions.