David-OConnor/stm32-hal #156: Add way to disable `defmt`

FeldrinH, 2025-09-25:
`stm32-hal2` depends on `defmt` to derive `defmt::Format` implementations. This is nice if you actually use `defmt`, but if you don't then the inclusion of `defmt` in the dependency tree causes confusion for tools like `probe-rs`, which detect `defmt` in your final binary and then fail because `defmt` is not correctly configured.

David-OConnor, 2025-09-25:
Concur; should be feature-gated

FeldrinH, 2025-09-25:
Thanks! I think it would also be good to remove `defmt-03` features from `embedded-hal` and `embedded-io`, because currently these include an outdated version of `defmt` if you enable `embedded-hal` or `embedded-io`. As far as I can tell the HAL doesn't actually need the `defmt-03` features.